fesch/CanZE #423: Climate - Max.°C gives value of 5°C ?

Albino-Zoe, 2017-02-02:
Climate - Max.°C gives value of 5°C ?
I have tested the manual temperature setting in the new screen. I noticed that the setting Max. °C gives a value of 5 ° !!! Also, when setting Min.°C
The other settings are displayed with the correct value. At the moment I have only a snapshot of manual operation 22 ° C - 26 ° C - Max. ° C - 26 ° C.

What's going on here???
[image]

yoh-there, 2017-02-02:
That is correct. Max and Min and Off are internally represented by these values. We need to discuss what to do with those.

@fesch what do you think? As we pass a SID to the graph function, this behavior was expected. IMHO we have a few options:
- do nothing ;-)
- remove setpoint from graph (not a great idea I think)
- use a virtual field that translates 0 to nothing (as in no graph. Is that even possible?), 5 -> 40 and 4 -> -10. (overkill??)

fesch, 2017-02-02:
Yes, I think a virtual field will be the easiest to implement here.

yoh-there, 2017-02-02:
Added and activated virtual field HeaterSetPoint (SID 800.6105.24) to use in climatech activity to avoid strange values for off / min / max. Thanks Alibo-Zoe. Committed in ClimatePower branch.

@FredLeudon As far as I am concerned this issue can be closed now but I am sure it also affects the iOS version if you have already implemented Clima. Would you be so kind please to close it once you are done OR if this is not relevant for you? Thanks!

FredLeudon, 2017-02-02:
Yes, same conclusion while I played with climatic activity yesterday.

I will do the same : add virtual field ...

FredLeudon, 2017-02-08:
Issue solved with 1.0.15 version available on Apple Store.

closed 2017-02-08 by yoh-there