fesch/CanZE #424: Reading data in the background while another screen is displayed

Albino-Zoe, 2017-02-02:
Since it is understandable that the screens can not be switched during driving, why not all values ​​are read in the background? 
Because if the opportunity for switching is given, then on the newly selected screen only from this time the values ​​are displayed. So I can currently select only one screen, although I am interested in the journey for several representations, e.g. Battery Module Temperature (Battery) and Range / State of Charge (Consumption).

------------------

Zoe R240 (08/2016), CanZE (01/2017), KW902 ELM327 (01/2017)

fesch, 2017-02-02:
The more fields are being red, the bigger the read intervals will be.

Reading fields in background is possible, but this feature has not yet been implemented. User should be able to select what fields they want to include and what fields not. This would imply big changes to all screens, but as other changes were pending, this has been pushed onto a later moment.

Albino-Zoe, 2017-02-02:
Clearly I understand :-)
Then it would be a good option for the future to leave to the user, which content he displays in the screen, thus adjustable.
As long as I have to understand larger effort connections
Because I have also adopted this great theme "CanZE" to understand the many unclear questions of functionality.
Also to understand probable mistakes, for which Renault is responsible (e.g. Functional heating after multiple pre-climate control)
Well, I'm looking forward to the future :-)

Albino-Zoe, 2017-03-08:
uups, I closed it, sry ... reopen !

fesch, 2017-03-08:
I started implementing the new logging feature yesterday ... so work is in progress.

yoh-there, 2019-01-13:
Closing this thread due to lack of activity

closed 2019-01-13 by yoh-there

Albino-Zoe, 2019-02-07:
@fesch: No motivation to complete this? I'm sad ...

yoh-there, 2019-02-09:
The issue was last commented on a year and a half ago. I am trying to keep the work-top-be-done a bit uncluttered. If nobody is picking up the issue further after all this time, I don't see the point to be honest.