fesch/CanZE #427: Test ELM327 Dongle - Error in test suite

Albino-Zoe, 2017-02-02:
FYI

[image]
---------------------
Zoe R240 (08/2016), CanZE (01/2017), KW902 ELM327 (01/2017)

yoh-there, 2017-02-02:
Confirmed. Thank you! 

Removed field from the Fields class, probably already a while ago. Will add back.

yoh-there, 2017-02-02:
Closing. Fixed in branch "development".

closed 2017-02-02 by yoh-there

Albino-Zoe, 2019-02-07:
Please open it again, the problem is in V1.29 with my R240 still or again!

yoh-there, 2019-02-09:
This is a stupid regression and fixed in the development branch. It should be pushed to the master branch (which represents releases soon). Reopening is not needed.