fesch/CanZE #429: Get stuck with Debug 5d7.0

StephanHi, 2017-02-11:
Using V1.21 build 2017.01.29@20:10 with a brand new MaxiScan 902 (KW902) on a Sony Xperia M2 (Android 5.1.1).

CanZE connects and reads Battery and Charging data from the main screen fine, but when I call up Charging from the technical screen it gets stuck with Debug 5d7.0 and subsequently no longer shows any data in any screen. Host LED on the MaxiScan flashes all the time, though. Reconnecting BT doesn't help. Restarting CanZE doesn't help. Switching MaxiScan off and on again doesn't help. After some time off it works again, but the behaviour can be reproduced.

fesch, 2017-02-11:
@yoh-there : it seams this dongle is being stuck in an ATMA. Maybe we should try to move completely away from the free frames and replace them, wherever possible with regular requests?

yoh-there, 2017-02-11:
I will try to make a translation table. If it works we might be able to
support 5 euro dongles ;-)

yoh-there, 2017-04-20:
No progress on this one (or anything CanZE to be honest. Short on time and need to MQTT-enable my charger, which is a serious project).

However, I think I got some (challenging) grip on the stability issues.
1) Since I expanded the wiring to my dongle, the dongle itself has lockups more often (hint: can't switch it off, so the processor is locked up). I suspect it is susceptible to sloppy power. I will try to add a beefy and a 100nF capacitor and see if it improves;
2) Android BT itself is sometimes unstable. The other day no connection possible, then I noticed R-Link didn't connect either. Then tried to switch BT off on the phone and even that was impossible. Might be my hardware (Cubot X15) or Android itself.

yoh-there, 2019-01-13:
Lockups are, unfortunately unavoidable. Reasons are:
- dongles locking up (see above);
- sloppy BT implementation on some Android phones. My own testing suggests this is due to the sharing of the radio hardware between WiFi and BT (a lot less problems when I switch off WiFi)

I am closing this issue as there is not more we can do other than we already do (you wouldn't believe the timeout detection, retry and reset code already in CanZe). We try to do better on the dongle side with the new DIY CanSee dongle.

Closing this issue.

closed 2019-01-13 by yoh-there