fesch/CanZE #431: Clima loop mode not working?

yoh-there, 2017-02-12:
Not sure if this is correct as I don't have an iOS device, but I saw a few screen prints of the CLIMA screen. There was no indication on the Clima loop mode line. If this is nonsense, please just close.

FredLeudon, 2017-02-13:
I will have a look ....

FredLeudon, 2017-03-04:
I had a look at the clima loop mode and it seems thad it is always "heather pump" mode. 

I switched from a/c on and off, de-icing, auto / manual with no change in clima loop mode. 

When I charged, clima loop mode was "idle"

What action make clima loop mode change for people who think it's working well?

Can somebody send a data logging file ?

yoh-there, 2017-03-04:
De-icing is not a choice. What you meant is de-misting. I think it is still
simply too cold for the AC mode to kick in (but maybe it is warmer where
you are).

What the field SHOULD indicate is how the coolant circuit is switched:
de-ice the main radiator, cool, or heat. My advise would be to leave it
like this until summer.

FredLeudon, 2017-03-04:
Ok,

There is perhaps a problem with my clima,

This winter, radiator etc were covered of ice....

Frédéric RICHARD

yoh-there, 2017-03-04:
There have been serious software updates t the climate computer. Maybe a
good idea to have that checked.

FredLeudon, 2017-03-15:
I tested ClimaTech activity and I assume Clim loop mode is working fine.

Closing this issue.

Feel free to reopen with screen shots if necessary.

closed 2017-03-15 by FredLeudon