fesch/CanZE #434: Energy consumed kWh

Albino-Zoe, 2017-02-15:
Hi,
this new value is not displayed with my R240.

Do you need logging or screenshots?

yoh-there, 2017-02-15:
I would appreciate a screeshot of the Charging tech and from Firmware,
after having tapped on the EVC. Thanks!

Albino-Zoe, 2017-02-15:
Electric Vehicle Controller
Diag version:06				
Supplier:001				
Software:0204				
Version:07C0
[image]

yoh-there, 2017-02-15:
I will need to ask this the community (neither @fesch  nor I own an R240). On my car it works fine, but I get it from the EVC and that box is way different on the R240

yoh-there, 2017-02-15:
I do not have the definition of that particular software drop, however, close one (6-001-0204-07700). In that definition, the field we display on that position does exist. So we have some confusion here.

yoh-there, 2017-02-15:
Confirmed by another R240 owner, so it seems motor type related. As I have no other up to date information, not much I can do.

Albino-Zoe, 2017-02-16:
Thx,

I hope from here   http://www.goingelectric.de/forum/renault-zoe-batterie-reichweite/durchschnittlicher-gesamtverbrauch-t21818-10.html   to get more information to resolve the issue ;-)

micc1, 2017-02-16:
Hi, I am new here and have been "directed" here from http://www.goingelectric.de/forum/renault-zoe-batterie-reichweite/durchschnittlicher-gesamtverbrauch-t21818-10.html.
I have an R240 from 05/2016 and have the same issue: energy consumed is not displayed. Can i help to fix this?

yoh-there, 2017-02-16:
Thank you micc1. What you might try to do is go to the DTC screen, select the EVC and do dump of all data. Then please drop the file here it created on your phone/tablet. It would help if you have a rough estimate of what your total consumption has been (in the order of your total km divided by 6)

micc1, 2017-02-18:
Hi yoh-there,

I am sorry, in my German version it is all different. I suppose I have 
to read out the error messages (Fehlerspeicher auslesen)? Here the 
choice "EVC" exists. However, I don't get any results, even after using 
the indicated procedure (enter key card, gear D, press Start for 5 
seconds...), it says "Keine Definitionen gefunden für diese ECU. Nur DTC 
Codes Anzeige.

Das DTC Feld exisitiert nicht."

yoh-there, 2017-02-18:
Ah, now THAT is a bug. I will look into that.

yoh-there, 2017-02-19:
@micc1 Can you please check your Android version and your CanZE?

@fesch do you have any idea? I can reproduce this on the play store version on an old Samsung S2 (4.1.2), but not the emulator or my normal phone (4.1.2) in the development version. Nothing has changed though since release in the DTC activity. Issue with dynamic class loading maybe, DtcActivity.java, lines 132-136.

micc1, 2017-02-19:
Android: 5.0
CanZE: 1.22

---------------------------------------------
Der perfekte Zeitpunkt seinen Gasversorger zu wechseln:
http://www.anpa.de/index.php?action=gasvergleich

yoh-there, 2017-02-19:
Thank you. @micc1 

I just installed development version on that old phone, and could not reproduce the problem. Then I re-installed from play store, and the problem is back. So it has something to do with the build, but I cannot figure out what.

fesch, 2017-02-19:
In that case I'll publish a new release to fix this.

fesch, 2017-02-19:
Merge to master is done, updated some lang files in order to be able to compile.
Release is done. Hope this works ;)

micc1, 2017-02-19:
Thank you very much for trying, but it seems to have made things worse: 
Now my CanZE 1.23 does not even connect anymore via Bluetooth (have 
tried several times, also after closing and/or deactivating bluetooth).

---------------------------------------------
Der perfekte Zeitpunkt seinen Gasversorger zu wechseln:
http://www.anpa.de/index.php?action=gasvergleich

yoh-there, 2017-02-21:
Any developments here? I know I am talking the "stupid" issues here but is your dongle paired in Android? And then selected in the Settings, and also using Type "Elm327"?

Albino-Zoe, 2017-02-21:
Hi again,

I have made the same way as micc1 3 days ago with the same results:
[image]
[image]

And now ?

micc1, 2017-02-21:
Hi, sorry, actually the dongle was not selected anymore, I can now 
connect with it. I have CanZE 1.24 now. But the error readout is still 
the same. Maybe i am making mistakes there. I follow the instruction 
(inset keycard, gear to D ...), but this is not very clear. Must I do 
anything else? Must the engine be switched on? The car tells me to take 
the key card out - must I do that? Afterwards I press the buttons "all 
data" and "get error messages" (or similar) in CanZE, but I don't get 
any lists/results/messages. The app produces a few lines of text, but 
they are not messages from the error count. If you want, I can translate 
and send them.

yoh-there, 2017-02-21:
We are talking three problems here in one issue. This is very hard to control and manage:

- no energy consumed for R240. This was the original problem and is for the moment unsolvable with a lack of EVC Diag files. For this reason I will Reopen the thread, and it is only for this purpose.
- Dongle possibly not paired or not coupled in Settings: Not taking this any further here. Please read the Q&A on the website.
- EVC DiagClass not found. This seems (not sure!) a build or a version error. I will open a new issue for this #442 and ask politely to continue there.

Albino-Zoe, 2017-02-21:
Thx
then I behave quietly and wait ;-)

yoh-there, 2017-02-21:
No lets keep this open. The original problem (total energy) would be nice to have if the R240 is able tot give it to us.

Albino-Zoe, 2017-02-22:
Sry, 

had my reading glasses set down and pressed the wrong button :-(

HarmOtten, 2017-03-07:
(CanZE Android version)
In my Zoe Q210 it shows like this:
"Energy consumed kWh = 9999.0"
which does not seem to be correct.

yoh-there, 2017-03-07:
Turns out the field is not working properly on the Q210 either. I will remove it. Issue closed, unfortunately

closed 2017-03-07 by yoh-there