fesch/CanZE #435: Screen charging - Battery kilometers is not always updated

Albino-Zoe, 2017-02-15:
Hi,
I already notice in V1.21 the value "battery kilometers is not always updated.
Excuse me for not telling it earlier, I was a little busy :-(
In V1.22 the same.
My test scenario:
At the beginning of the drive I start CanZE.
I switch between all screens and make a screenshot.
Then I decide for one screen and begin my drive.
At the finisch I switch again all screens and make screenshots.
In the screen "Charging" all values are correct, only "battery kilometers" still has the value from the start of the car trip
Do you need logging or screenshots ?

yoh-there, 2017-02-15:
No screenshots needed. I will use your routine and see if I can reproduce it.

yoh-there, 2017-02-16:
I am closing this issue. It seems by design (not mine, but Renaults). I drove for a bit at the battery km stayed static. When I had locked the car the current odo km was transferred to the battery computer.

closed 2017-02-16 by yoh-there

Albino-Zoe, 2019-02-07:
Closed is accepted ;-)

yoh-there, 2019-02-09:
Thank you sir