fesch/CanZE #436: Climate - Synchronization/time problem ?

Albino-Zoe, 2017-02-15:
Hi, again ;-)

My test scenario ...You know it by now ;-)
At the beginning of the drive I start CanZE.
I switch between all screens and make a screenshot.
Then I decide for one screen and begin my drive.
At the finisch I switch again all screens and make screenshots.

[image]

In this test, the time axis was not the same at the end.

I also noticed that values ​​are not so regularly present, mostly "Internal (°C)".

----
Zoe R240 (08/2016), CanZE V1.22, KW902 ELM327 (01/2017)

yoh-there, 2017-02-15:
The Clima computer seems slow, and the graphs are separate entities, IOW, they are horizontally shifted as soon as a value comes in and, so what you are seeing is the result of the same problem. There is not much we can do about the responsiveness of the CLIMA computer I am afraid.

In my car (Q210) it seems it is mostly the RPM lagging.

I will leave this issue open for a few weeks to give others a chance to comment. At the moment I can't think of a reasonable change to "fix" it though. Thank you for reporting though!

Albino-Zoe, 2017-02-15:
Thx 

I just found that: http://canze.fisch.lu/qa/#comment-765

(Loc Dao) 2. As far as I know, the fields log will log the current visible screen, if we switch to another screen, it will stop those parameters from that screen and the fields.log continue to log the other parameters of the new screen we switch to, is it possible to log multiple screens (or all screens) at the same time? By this way we can have many parameters to observe with the same time duration.

Re: (Bob Fish) Not yet. A customizable logging screen has been planned but not yet implemented. Will be probably in the next release.

This not solves the problem ???
I'm not sure if I understand the facts correctly, I'm not a native speaker .... google-translater is my friend ;-) and I can not program, even "hello world" :-))

yoh-there, 2017-02-15:
Don't worry about language. Neither of us is a native speaker.

No it will not solve the problem. The queuing mechanism is quite complex. Some fields are retrieved as fast as we can (all in the climate screen!), some only slowly, i.e. SOC, SOH, as it doesn't make sense at all to query them very often, some are queried independent of the activity shown, i.e. speed for safe driving control.

yoh-there, 2019-01-13:
Closing due to lack of activity.

closed 2019-01-13 by yoh-there

Albino-Zoe, 2019-02-07:
It was not lack of activity of mine, I was hoping that the behavior will improve somewhat at some point in time behavior. But did not happen until the current update V1.29.
Closed is accepted ;-)

yoh-there, 2019-02-09:
I hope you follow the blog. The DIY dongle is heaven compared to a cheap ELM clone. Having said that, there is a secret plan to improving even the ELM control. It is a lot of painstaking work though, compared to which the other issue, still open, about nomenclature is a joke. But we'll get it done.