fesch/CanZE #437: Climate - Development in the future?

Albino-Zoe, 2017-02-15:
I think the screen "Climate" is now very good :thumb

Are you also working to display the remaining values ​​here?

- The 3 buttons for the blower (Passenger compartment air distribution
adjustment.)
- The button "AC off" (Air-conditioning deactivation control)
- the button "Air-conditioning deactivation control"
- ...

You know for sure what I mean ;-)

I think it is very important to have a graphical display which also shows the progress of the following values:

- Engine fan speed 
- Climate loop mode
- External (°C)
- Internal (°C)
- Setpoint (°C)
- Automatic mode (on/off) 
- Air-conditioning deactivation control (on/off)
- see No.2 in snapshot -> energy consumption of accessories (air-conditioning, radio, etc) -> Is it really "radio, etc" or only air-conditioning ?
[image]

Because the heating in winter especially strongly determines the consumption or an e-car driver must freeze ;-)
Enough for the moment ...
:-)

----
Zoe R240 (08/2016), CanZE V1.22, KW902 ELM327 (01/2017)

yoh-there, 2017-02-15:
There is no plan to include all those field. We are thinking about adding a line "ECO mode". The middle graph was meant to show kW, but that value was very unpredictable. RPM seems more stable and I think that gives most information people can take in: More RED = more energy usage from the heater/AC ;-)

Albino-Zoe, 2017-02-15:
Last question ;-)

Can you show whether the heat comes:

- only from the heat pump
- only from den "PTC"
- from both

Or there are no single values ​​for this?

yoh-there, 2017-02-15:
To be honest, we haven't dug into that. The only thing I know is that on normal ZOEs it is on the 12 volt bus, meaning it can never be more than 400 watt, probably a lot less. Compared to the heat-pump that is minute. For the Nordic versions the mechanic manual says it is on the 400 volt bus and there is an unused 400 volt connector on the junction box, so that could be true. We don't have more details, sorry.

Albino-Zoe, 2017-02-16:
Thx ... then I have to change my expectations to CanZE :-(
Is not your fault .... But I remain hopeful ;-)

yoh-there, 2017-02-16:
I had a quick look in the documentation. Seems like my 2013 Q210 has an 1800 Watt PTC :-o I honestly have no clue how that should be powered. AFAIK the DCDC converter is 1kW. Maybe it is only used with the DCDC converter in full power setting anf then taking another 50 amps from the battery. Wow, I have a hard time believing this, but if true, I understand better why the battery is on a 3 year replacement schedule.

So, I will keep this item open as it seems we can do it.

serbanm33, 2019-01-28:
Hello,
I have searched for a place to give you the information that I have regarding the battery heating and I think this is a good place.
Basically ZOE (I have an R90, 41kW produced in 2018) can be fitted with two types of battery heaters:
- HV PTC  - You can check if you have this with DDT4All in "HVAC->X10->PTC Cabin & Battery Mode X10-> Option:PTC HV Cabin" or "EVC -> Vehicule -> Configuration -> Option:PTC Config V2" - I do not have this; I have tested a Q90 and an R240 and both cars do not have this so I asume Renault is equipping cars with HV PTCs only in some markets or not at all. 
- 14V resistor controlled by two relays. Some people say that the 14V resistor has a few steps (0W - no relay, 333W - relay 1, 600W - relay 2 and 1000W - relay 1&2) and the two relays controls what steps are activated - My car has this; The control for this seems to be in BCM ECU -> CLIM - PTC -> CLIM - PTC - Configuration. I tried to understand this data but for the moment it not clear at what bat temp the relays are activated further investigation is required.

What I managed to do is to start battery heating on demand on some cases:
- Schedule the charging to start some time in the future (best setting: two hours in the future)
- Connect the car to a charging station - can be any station, even a portable charger, important is to be able to provide around 2kWh
- The HVAC fan will start at a lower speed setting, you will hear the fan in the back of the car. In CanZE in Climate screen, HV Bat conditioning mode will be "Heat cond req"
- This will work for around 2h or until the battery temp is ~12 degrees Celsius but not for more then 2h
- In the best case I have observed until now the battery is heated with 7-8 degrees per hour
- The consumption will not be constant, just some spikes and total consumption for 2h is between 1.7 to 3kW
- The real SOC will remain almost the same and usable SOC will increase when the bat temp is increasing.
 
Monitoring the consumption on 14V circuit I noticed a 40A or a 7-8A consumption. 40A seems to be a step on the resistor (600W) and the fan, and 7-8A just the fan. I think the car decides what to use based on SOC, bat temp and external temp. On my tests the 40A consumption is visible only for lower SOC (less then 50%) and negative external temps.
The modules seems to heat up at different rate, the center modules are heating up more quickly (this was to be expected) and at the end of heating the deference between modules can be up to 6-7 degrees. I do not know how safe is this for the battery.
If you monitor in CanZE in climate screen, the HV Evaporation temp will slowly increase. 
I have also noticed that the battery heating is active for a few minutes during charging (40A consumption on 14V circuit) if HV bat temp is less then 8,9 degrees.
Heating the battery before you start driving has an impact on the consumption, a huge impact on charging speed and bat temp determine how much SOC is reserved (in some cases at low bat temps I noticed that even 10% is reserved).

If will be possible it will be good to have an option in the future to heat up the battery to a settable temperature (10 degrees should be ok). Renault already received this request from me but I do not think they are rely considering implementing this so best hope is for CanZE :)

yoh-there, 2019-01-28:
Brilliant info, thank you! Neither Bob, nor myself own a car with a 41 kW battery, so there is not much we can do or verify or implement reliably, but a lot of people are dying for this.

serbanm33, 2019-01-29:
Thank you very much, but the credit is not fully mine. This started after my charging rate doped drastically in the winter and an issues that Renault is not willing to fix: ZOE charges even when HV bat temp is bellow 0 (in may case even at -3 degrees); this will destroy the battery and the SOH will drop like hell if you allow this to happen. So after reading some posts on [;](https://www.goingelectric.de/forum/renault-zoe/) I just did the testing and collected the data.   
Anyway, glad I can help and if you require any other data regarding this or any testing on R90 41 kW I can try to help.
Regarding the 1800W PTC, this is for the pasager cabin. At this point I think this is actually an HV PTC but more testing is require.

Thank you for giving us access to data that Renault is hiding about our cars.

yoh-there, 2019-01-30:
I am not so sure about the destruction. Renault has rented out a quite a few millions worth of car batteries, also the 41 kWh ones and they have a habit of fiercely protecting what is theirs. The older batteries are very capable of charging at low temperatures albeit also at lower rates.

The main complaint about the 41 batteries is their high power charging optimum is in a far more limited region than the older ones indeed.

The 1800 Watt cabin PTC is, AFAIK, a HV version only present in the Nordic versions. The non-Nordic version has, again AFAIK, a 14 volt 250 Watt version. Probably more for windscreen defrosting than for comfort!

serbanm33, 2019-01-30:
Some details for the 41kW version: [Here](https://drive.google.com/file/d/1_DRl_6c5hbxCMZrxztnu7Pbnc8h3sYVC/view) (Renault ZOE Owners Club). I'm not entirely shore this is the case for all markets. In my case I thing there is only one 14V resistor for battery heating (600W) or this is the only one for which I was able to see consumption.
The deference between 22kW and 41kW  seems to be the pipes for HV battery conditioning system. It is possible that only the nordic countries have the cabin PTC but in my case this it is configured (HVAC->X10->PTC Cabin & Battery Mode X10-> Option:PTC LV Cabin has value 1800W) .

Regarding the low temperature charging, the BMS for 41kW seems to be much more active in my opinion. The estimated range is more conservative and if you drive more sportive the range is affected much faster then for 22kW; the same is applicable if you kip the car started for a long time.

What is very strange is that Renault is allowing charge for 41kW version even when the battery temp is bellow zero. I did this only once (fortunately) but considering what we know about lithium batteries I do not recommend this for any version.

yoh-there, 2019-02-09:
I have added the PTC relays to the CLIMA screen. Now closing the issue. Thank you again for the information, and feel free to reopen.

closed 2019-02-09 by yoh-there

Sterillium, 2021-11-06:
Was the PTC status removed from the CanZE Climate screen? I can't seem to find it (Q90, 2018).

yoh-there, 2021-11-07:
Yes. AFAIK these fiels alwayes reported "off" and thus were basically useless. If anyone can provide more reliable info, ie through DDT4ALL, happy to consider putting it back in of course.