fesch/CanZE #438: Add resistance values to technical information.

sordoneF, 2017-02-16:
In the Diag codes are several resistor values. Can you add those to the (a) technical screen ?
app/src/main/java/lu/fisch/canze/actors/EcuDiagBCB.java 
392	                        +"793,24,39,1,0,0,Ohm,225062,625062,ff,Mains ground resistance\n" //

app/src/main/java/lu/fisch/canze/actors/EcuDiagLBC.java 
48	                        +"7bb,360,375,100,0,0,ohm,2101,6101,ff,21_01_#46_HVIsolationImpedance\n" //
1191	                        +"7bb,24,39,1000,0,0,Ohm,221411,621411,ff,1411_Isolation Impedance\n" //

app/src/main/java/lu/fisch/canze/actors/EcuDiagEVC.java 
479	                        +"7ec,24,39,100,0,0,Ohm,2233EE,6233EE,ff,Resistance insulation value\n" //

When charging I see some go to 1MOhm but in some error situations this can be different.
And might be a reason for an error in charging.

yoh-there, 2017-02-16:
Sordonne, have you read the reason for the kick starter funding in the Dutch forum? We can't get the BCB open. How do you see those values? Do you have an R240 (maybe that one is talkative).

All the requests you make are programmed, including harmonics. They are waiting for this problem.

sordoneF, 2017-02-16:
[image]
The values I found in the source of Canze on Github (search for ohm). I also see one of the values in Canze4iOS in EVC Diag on my Q210, see picture above. I did not check the LBC.
When the values are low when they should be high, it could indicate a problem.

yoh-there, 2017-02-17:
Personally, I think the dump file that is created when you pull up this screen is enough for those kind of diagnostics. What do you think @fesch ?

fesch, 2017-02-17:
yes, I agree on that. Techies will find in there whatever they need.

sordoneF, 2017-02-18:
Hmm. If you follow that reasoning, you can remove the charging screen since it is all in the Diag screen. I think the resistance values could influence the charging behavior so it would make sense to show it in the "charging (technical data)" screen.

yoh-there, 2017-02-18:
Isolation resistance does not influence charging behavior. It is a diagnostic of serious problems with the wiring of the car. We're into serious problem territory here.

As for putting it on the screens, two things:
1) [and I know this is an old argument]: this is an open source project. If you feel this is very important (which we obviously disagree with),

yoh-there, 2017-02-28:
I am closing this issue. We are working on displaying as much information as we possibly can on charging failures, as we know ZOE is picky on what she's been fed. I hope that general statement will cover this issue sufficiently.

Deep in-car diagnostics need to be done through DTC's and parameter dumps and we're working on those definitions; we're not trying to recreate the dealer's diagnostics tool. Internal resistance values fall in this category, IMHO.

However, we will continue to try to identify and display things that the driver can slightly influence, report on, or help understanding what's going on. To illustrate this: ground resistance errors, which can be a cause of charging errors should be reported to the chargepoint operator, and is an indication to the driver he should avoid this chargepoint, we WILL try to display.

closed 2017-02-28 by yoh-there