fesch/CanZE #439: Can not get Canze to work on IOS

exycutez, 2017-02-18:
Hello,

I just got a wifi obd from china (http://s.aliexpress.com/iIZNjQzi). As i read on the canze website, any v1.5 wifi obd should work, however it just says "connecting to wifi-socket" and it says that forever. I have to quit the app to get the message away. I have been using the static ip and subnet, and default port settings, but still no luck..

Any thoughts? I have now ordered a new kw902 wifi dongle, as i assume ita the dongle?

GertjanVB, 2017-02-18:
The dongle you're indicating is working for me. Did you connect WiFi with the password? You need to do that only once. First time connections seems to take a while. Is the automatic connection slider in "settings" active? Can you see an active WOBD WiFi connection in iOS settings?

yoh-there, 2017-02-18:
Thanks GJ. I would appreciate if someone would write an article for the website on what works for iOS and what not and how, like for the Android version (preferably better ;-) ).

GertjanVB, 2017-02-18:
I will have a try...

exycutez, 2017-02-18:
There was no password to connect to wifi..

This is what i have done:

1. Plug in the dongle (light is blinking some different colors, and then it is showing a solid red led)
2. Connecting to the wifi obd, i recieve ip 192.168.0.0X (from 1-9), and the router ip is 192.168.0.10.. then i set static ip to the same as the regular ip (also tried 192.168.0.123) and subnet to 255.255.255.0.. 
3. When this is done, i hit "try find dongle" and it says it might be the adress 192.168.0.10, so i dont mess around with that in canze.. and i i try, i get error..
4. Trying to connect, and it says "connecting to wifi-socket", longest time i have been waiting on it is 1 hour!

GertjanVB, 2017-02-18:
When no connection with the dongle is made by CanZE, you get a "unable to connect" message
Which CanZE version are you using, mine is 1.0.15.7
Do you have an active wifi connection in iOS with the dongle,

exycutez, 2017-02-18:
If i try putting "192.168.0.10" as static ip on wifi settings on iphone, i get unable to connect bla bla.. if i changw ip field in canze i get unable to connect.. 
Newest canze version (1.0.15.7), i did not drive the car, the car was charging and closed for most of the time, however, i have tried to have it connected while driving for approx 15 min, is that long enough? Car settings is r240, have not changed the socket read timeout.. get unable to connect when no dongle in car. It dowsnt freeze with the dongle, it just load forever with the attempting to connect to wifi socket..

Phone is iphone 7 plus, same dongle as on ali.

exycutez, 2017-02-18:
By the way, i just saw that the wifi symbol on my screen up left corner is missing when i enter the static ip..

However, it says im connected to the dongle, and i am getting an ip from it, if i trust canze when hitting "try find dongle"

GertjanVB, 2017-02-18:
There is no wifi icon in the top left during connection with the dongle, that's ok
CanZE seems to recognize the dongle, but there seems to be no connection and data transfer 
I can't remember the exact procedure for connecting, but as a result there's a wifi WOBD connection 
I was convinced you had to use the password 12345678 to make this connection, which is made outside CanZE in the iOS setting panel.
Tomorrow I will erase my settings and go through the procedure to get connected.

exycutez, 2017-02-18:
It is not passeord protected for me, just plug and connect! Just tried connecting for 35 mins while driving, when stopping the car CsnZE still tried to connect to it!

exycutez, 2017-02-18:
Well, i bought 2 of theese cheap dongles, and found out that it was this (http://s.aliexpress.com/V7FbeyUv) that i have now and that has arrived.. it says "obdii interface" instead of "ELM327" interface!

Could this be the problem?

FredLeudon, 2017-02-18:
CanZE for ios uses a Wifi ODB2 dongle.

It means that there are two connexion to perform. 

First step is to join dongle's wifi network.
In some case, dongle use open wifi network (mine is called wifi-odb2) and in other cases, dongle provides à private wifi network (so You have to know wifi password)
- go to Apple iPhone settings, wifi and join your dongle wifi network.
- if successful, tap the i sign to view ip, subnet and gateway assigned by the dongle's dhcp server. 
- laungh CanZE, go to settings. Enter gateway as dongle's ip address
- all wifi dongle I tested uses port 35000
- slide to connexion page and tap connect, normaly, it works fine.

To check Your Wifi dongle, You can install this software (EOBD Facile - OBD2 Diagnostic Automobile par OUTILS OBD FACILE SARL
https://appsto.re/fr/HqzJQ.i)

exycutez, 2017-02-20:
I tried the OEBD-Facile app that was recomended above, took 2 secounds for it to find the dongle and it started blinking!

CanZE cant find it it seems, wierd? How long does it usually take for you guys for CanZE to find the dongle (first time)?

GertjanVB, 2017-02-20:
This is a concept guide for installation, hope this is helpfull;
[CanZE iOS Download and installation guide.pdf](https://github.com/fesch/CanZE/files/786904/CanZE.iOS.Download.and.installation.guide.pdf)

fesch, 2017-02-20:
@GertjanVB : May I post this guide on the CanZE Site as well?

GertjanVB, 2017-02-20:
Sure, Frederic and Jeroen have viewed and commented this document yesterday evening. Jeroen is in the proces of getting it on your site. Don't know what the routing exactly is

exycutez, 2017-02-20:
Well, this is what i have been doing, except the router and dns part. Still not working..

I assume its one of two:
1. CanZE does not work on i7 plus
2. CanZE dows bot work with an obdii wifi dobgle without pw protection

GertjanVB, 2017-02-20:
Did you start form scratch? eg. flush data on CanZE, close CanZE, remove and forget the WifiODB network and connect again before starting CanZE.
My CanZE runs on a 4S, can't tell wether or not there are problems with a 7plus

exycutez, 2017-02-20:
I tried flushing, reinstalling, forgetting wifi and so on..
Waiting 15-20 minutes on the attempting to connect to socket.. now im out of ideas!

Maybe the developer can answer regarding if its working with a open network obdII?

FredLeudon, 2017-02-20:
Sorry if CanZE don't work well on iPhone 7.

I tested it on iPhone 4s, 5c and 6

I will test it with iPhone 7 on simulator to try to reproduce your problem. 

Regards

Frédéric RICHARD

FredLeudon, 2017-02-20:
I just launched a test on iPhone 7 Plus simulator and it worked perfectly. 

Could you please send screenshots from:

iPhone settings app / wifi and wifi details

CanZE settings tab

Regards 

Frédéric RICHARD

exycutez, 2017-02-20:
Thanks for your effort to help,
I will send you screenshots when i get to the car later today!

exycutez, 2017-02-20:
[image]
[image]
[image]

FredLeudon, 2017-02-20:
Ok, it seems correct. 

Try to uncheck "safe driving mode" and say if it's working better please.

You can try to uncheck "automatic connexion" and after launching CanZE, tap to the "connect" button 

Normally, connexion is made in a couple of seconds. 

Frédéric RICHARD

exycutez, 2017-02-20:
Troed unchecking those two.. Im close to giving up :/

Wierd that it works so fast with the other app u suggested! Do you think its because its an open wifi with no password?

FredLeudon, 2017-02-20:
No, that not the reason why it doesn't work. 

Assume "debug log" is on.
 
On main screen, tap delete log files button 

Force CanZE to quit by double pressing home button and sliding CanZE up

Launch CanZE, wait a moment and go to main screen ant tap "send feedback"

Mail app will open with a new mail for me. 
Delete attached file data.fic witch I don't need, send. 

I hope I can see whit it's not working in log file. 

A new version of CanZE ave been sent today to apple. It's pending for verification.

Perhaps installing this new version will help (I upgraded développement software) to solve your problems. 

Frédéric RICHARD

sordoneF, 2017-02-20:
I have an iPhone7 (Model MN932ZD/A with iOS 10.2.1) and it works well with Canze build 1.0.15.7 and my Wifi dongle. It has them same settings and no password, connect takes ca. 2 seconds (after selecting wifi for OBD in Wifi settings)

FredLeudon, 2017-02-20:
Ok, if CanZE works with iPhone 7, perhaps is it the dongle the program can't talk to?

Would you agree to send me your dongle, so I can test it and eventually (if possible) upgrade CanZE code to fit with your dongle?

Frédéric RICHARD

GertjanVB, 2017-02-20:
I thought It's the same dongle I send you earlier, you tested it and it was OK. May be a different version. Exycutez, can you confirm?
[image]

FredLeudon, 2017-02-20:
Sorry but I don't remember ;-)

Frédéric RICHARD

fesch, 2017-02-20:
Looks exactly like the one I have too, which is working ... 
I will test tomorrow with the latest version.

GertjanVB, 2017-02-20:
@FredLeudon : never mind ;-)  it was early september last year. I bought the dongle here http://www.ebay.nl/itm/ELM327-Wifi-Interface-Diagnostique-Auto-ODB2-Detecteur-Panne-iPhone-Android-/222020430447?hash=item33b172ea6f:g:OBsAAOxy6MBSRGBq

yoh-there, 2017-02-20:
Please be aware these are VERY generic dongle cases, including the labeling. Inside, these can be vastly different.

exycutez, 2017-02-20:
I can send you the dongle, but i am waiting for another one that arrives this week! Will try that one before sending it to you, since i eill recieve the new one before you get the one i have now!

Sending you feedback file now!

FredLeudon, 2017-02-23:
I didn't see anything in log files. 

The best way is to wait for your new dongle. 

If it works with new dongle, could you send the old one to me?

exycutez, 2017-02-23:
Allright, i will send you the old one if the new one is working.

Actually, i got 2 that is the same as my current one arriving verry soon, and a kw902 wifi that arrives abit later.. The Konwei kw902 should atleast work?

exycutez, 2017-02-23:
Secound dongle arrived (have one more of the same as we have spoken about, and one kw902 arriving also), and tested it with the new version of Canze.

This is what im getting now, it says the same after 5 minutes.
[image]

So, is this normal? Now it doesnt even try to connect to the dongle it seems?

FredLeudon, 2017-02-23:
Hi, normally a couple of seconds. 

There is something wrong but I'm not sure that it is due to CanZE.

Can you again send screenshots of wifi settings and CanZE settings. 

For testing, assure you are using dhcp in iPhone's wifi settings.

Is your dongle recognized by eodb facile ?

If yes, go to "calculateurs", "console" and type ATZ and "envoyer" and send a screenshot.

Frédéric RICHARD

exycutez, 2017-03-08:
Tried 3 different dongles, like the one i have linked to.. no one is working..
It seems like they dont work with OEBD-Facile either, so my previous answer with it working with OEBD-Facile (the app you recomended for testing) is not correct. In OEBD-Facile, all the dongles have the led's blinking, but it is not connecting.

Seems like the dongles are a cheap/fake obd ii? Now i am waiting for a kw902 wifi dongle to arrive (should arrove in some days). If that is not working, i would ask for any of you, that have a dobgle that works, to send me a link to one that is working?

yoh-there, 2017-03-08:
I propose to close this issue. Sorry if this sounds "Dutch", but we've put warnings about dongles and which ones work all over the website and several fora. We have no stake in the KONNWEI's, but it's the **only** one we know it works without too many problems **and** that can be ordered with limited risk of getting a non-working knock-off. Yes, we do see other successful dongles, but always no-brand and the next one can (and often will) fail.

For my co-developers: harsh as it may sound, my unasked-for advise is to not put any effort in it. It is a sinkhole of precious time and energy.

For the issuers: get a KONNWEI. If it fails, feel free to report.

exycutez, 2017-03-08:
Yeah, that is all fine by me.
I read somewere that any 1.5 wifi dongle worked, and i have seen pictures of other users dongle, so i thought it would work!
I am now waiting for the kw902, so lets hope it will works!

Thanks

closed 2019-01-06 by yoh-there