fesch/CanZE #441: CanZe exits after settings

paprmakr78, 2017-02-20:
Once I've called the settings, the app exits and I need to reinstall it. Android Version 1.21 and newer.  Updates do not bring any improvement. Before the update to 1.21 it still works. As soon as I save the settings, CanZE quits. Sony XPERIA Z1C, Samsung Galaxy Tab 2 10.1 WiFi and others.

fesch, 2017-02-20:
Maybe you could try to install, then goto settings and check the "debug log". All tough the application quits, it should write something to the log. You could then post it here.

paprmakr78, 2017-02-21:
2017-02-21 05:50:23.942: Ecu does not exist:LINSCH
2017-02-21 05:50:23.968: 130..44
2017-02-21 05:50:23.971: 186..16
2017-02-21 05:50:23.972: 186..40
2017-02-21 05:50:23.984: 18a..27
2017-02-21 05:50:23.985: 1f8..16
2017-02-21 05:50:23.986: 1f8..28
2017-02-21 05:50:23.987: 1f8..40
2017-02-21 05:50:23.988: 1fd..0
2017-02-21 05:50:23.989: 1fd..48
2017-02-21 05:50:23.990: 427..40
2017-02-21 05:50:23.991: 427..49
2017-02-21 05:50:23.992: 42a..30
2017-02-21 05:50:23.996: 42a..48
2017-02-21 05:50:24.000: 42e..0
2017-02-21 05:50:24.001: 42e..20
2017-02-21 05:50:24.002: 42e..38
2017-02-21 05:50:24.004: 42e..44
2017-02-21 05:50:24.005: 42e..56
2017-02-21 05:50:24.006: 430..24
2017-02-21 05:50:24.007: 430..38
2017-02-21 05:50:24.008: 430..40
2017-02-21 05:50:24.011: 432..36
2017-02-21 05:50:24.013: 4f8..4
2017-02-21 05:50:24.018: 5d7..0
2017-02-21 05:50:24.019: 5da..0
2017-02-21 05:50:24.019: 654..2
2017-02-21 05:50:24.020: 654..25
2017-02-21 05:50:24.022: 654..32
2017-02-21 05:50:24.023: 654..42
2017-02-21 05:50:24.029: 656..48
2017-02-21 05:50:24.035: 658..33
2017-02-21 05:50:24.037: 65b..41
2017-02-21 05:50:24.039: 673..0
2017-02-21 05:50:24.040: 673..2
2017-02-21 05:50:24.042: 673..5
2017-02-21 05:50:24.044: 673..8
2017-02-21 05:50:24.046: 673..11
2017-02-21 05:50:24.047: 673..16
2017-02-21 05:50:24.049: 673..24
2017-02-21 05:50:24.051: 673..32
2017-02-21 05:50:24.052: 673..40
2017-02-21 05:50:24.054: 699..8
2017-02-21 05:50:24.056: 760.6180.56
2017-02-21 05:50:24.058: 760.6180.64
2017-02-21 05:50:24.059: 760.6180.128
2017-02-21 05:50:24.061: 760.6180.144
2017-02-21 05:50:24.063: 762.6180.56
2017-02-21 05:50:24.065: 762.6180.64
2017-02-21 05:50:24.066: 762.6180.128
2017-02-21 05:50:24.067: 762.6180.144
2017-02-21 05:50:24.069: 763.6180.56
2017-02-21 05:50:24.070: 763.6180.64
2017-02-21 05:50:24.071: 763.6180.128
2017-02-21 05:50:24.072: 763.6180.144
2017-02-21 05:50:24.073: 764.6121.26
2017-02-21 05:50:24.075: 764.6121.36
2017-02-21 05:50:24.076: 764.6143.86
2017-02-21 05:50:24.077: 764.6143.110
2017-02-21 05:50:24.079: 764.6143.134
2017-02-21 05:50:24.080: 764.6144.107
2017-02-21 05:50:24.081: 764.6167.40
2017-02-21 05:50:24.083: 764.6180.56
2017-02-21 05:50:24.084: 764.6180.64
2017-02-21 05:50:24.086: 764.6180.128
2017-02-21 05:50:24.087: 764.6180.144
2017-02-21 05:50:24.088: 765.6180.56
2017-02-21 05:50:24.089: 765.6180.64
2017-02-21 05:50:24.090: 765.6180.128
2017-02-21 05:50:24.091: 765.6180.144
[... 582 lines cut ...]
2017-02-21 05:50:33.271: BluetoothManager: Closed
2017-02-21 05:50:33.266: BluetoothManager: Starting new try
2017-02-21 05:50:33.283: MainActivity: onPause
2017-02-21 05:50:33.286: MainActivity: onPause > leaveBluetoothOn = false
2017-02-21 05:50:33.288: Device: clearFields
2017-02-21 05:50:33.292: MainActivity: stopping BT
2017-02-21 05:50:33.296: MainActivity: stopBluetooth > stopAndJoin
2017-02-21 05:50:33.299: Device: stopping poller
2017-02-21 05:50:33.302: Device: waiting for poller to be stopped
2017-02-21 05:50:33.304: Device: >>>>>>> pollerThread is NULL!!!
2017-02-21 05:50:33.306: Device: pollerThread joined
2017-02-21 05:50:33.307: Device: poller stopped
2017-02-21 05:50:33.309: Device: clearFields
2017-02-21 05:50:33.313: MainActivity: stopBluetooth > BT disconnect
2017-02-21 05:50:33.328: BluetoothManager: Waiting for retry-thread to stop ...
2017-02-21 05:50:35.281: BluetoothManager: Closing socket
2017-02-21 05:50:35.305: BluetoothManager: Closed
2017-02-21 05:50:35.360: SELECT: deviceAddress = 00:25:CB:4A:A0:05
2017-02-21 05:50:35.362: SELECT: deviceName = tanJack BT r2a6i
2017-02-21 05:50:46.407: Settings.deviceAddress = 00:10:22:09:03:30
2017-02-21 05:50:46.409: Settings.deviceName = CHX
2017-02-21 05:50:46.441: MainActivity: onActivityResult
2017-02-21 05:50:46.443: MainActivity: onActivityResult > requestCode = 7
2017-02-21 05:50:46.444: MainActivity: onActivityResult > resultCode = 0
2017-02-21 05:50:46.445: MainActivity: loadSettings

fesch, 2017-02-21:
Thanks to the log file, I was able to find the code which seams to produce some exception, but there are a few lines that could have an impact. I thus recompiled a new version which gives - hopefully - more about in that section. You can find it here and then, probably, do the same operation and post that log too:

http://canze.fisch.lu/uploads/2017-02-21-canze.apk

paprmakr78, 2017-02-21:
The app I have installed and tried. Now, the app stops, but I can not connect to the adapter. Also the setting of the vehicle (with me R240) is not saved, remains with Q210.
[debug-2017-02-21 13-59-29.460.log.txt](https://github.com/fesch/CanZE/files/790170/debug-2017-02-21.13-59-29.460.log.txt)

fesch, 2017-02-21:
... but the log contains the needed details about the error :-)

Could you try this version?

http://canze.fisch.lu/uploads/2017-02-21-canze-2.apk

paprmakr78, 2017-02-21:
Now I have tried the last version and it all works well. Only the response times are slow, but this is probably due to the logging. Thanks a lot for your quick help.
[debug-2017-02-21 14-27-02.907.log.txt](https://github.com/fesch/CanZE/files/790229/debug-2017-02-21.14-27-02.907.log.txt)

closed 2017-02-21 by fesch