fesch/CanZE #443: Language selection option

zmeglic, 2017-02-23:
It seems currently the language in CanZE is  directly selected by the android language setting.

I suggest a simple setting that can override this. It would make it much easier to debug also, since you could always get the reports in english.

yoh-there, 2017-02-23:
@fesch is this easy, given the locales? I know string selection in code is pretty concentrated in MainActivity, so any change can probably be concentrated there, but I don't know about the rendering of the layouts. Do we want to do it?

DiLeGreen, 2017-02-24:
me to. 
I personally hate german translations of technical terms, i suggest to add a dropdown menu or a "Show in English language" checkbox to select the language in the settings.

yoh-there, 2017-02-28:
Assigning you as the Settings & Android specialist ;-)

wolfijenne, 2017-03-01:
I'd love this option too!

yoh-there, 2019-01-13:
Bumping this issue. I know we do a lot of things on the set locale, so it might be too much work.

My personal stubborn opinion is that the current solution is right. It's "not fair" to have developers put effort in your preference to set your phone to language A (as that seems to be the better choice overall), but then wanting to make an exception for app X and Y and run them in language B. If you like B, set your phone to it ;-) BTW it is what I do. My phone is on English while my native tongue certainly is not.

yoh-there, 2019-01-28:
Closing this issue due to lack of follow up

closed 2019-01-28 by yoh-there