fesch/CanZE #444: Screen Request

wolfijenne, 2017-02-24:
This would be the Drive Screen without Information already on the car's display. (Maybe an additional Screen).

I find it disturbing that the car's display does not show the SOC level but only the remaining distance which basically means not so much.
As living in the mountains, the breaking/regeneration is important. Therefore, I would love to see these things:

– Battery Temp
– Power
– Max Regen Power
– Mechanical Brake Power
– % SOC
– (Speed)

The values in brackets are also available on the car's display and thus not so important, however, the Power value is important as it goes toegether with the brake power. The brake power would be good to have in kW and not in Nm as the Power value leads to wear off and waste of energy.

yoh-there, 2017-02-25:
Isn't this what is exactly in the consumption screen? (please also note that in this screen there is also the blue bar), see also #445

wolfijenne, 2017-02-25:
Yes and No. The information is almost there (except for mechanical Brake, Max Genereration Power and Batt Temp), however, during driving, charts can and should not be read. There are only big numbers possible.

Despite the discussion under #445 I can't see the blue aiming bar below in the driving screen as in the screen shot on the website. I am using 1.24 on a Huawei G8 with a R90.
[image]
[image]

wolfijenne, 2017-02-25:
Correction... I found the blue bar :)

yoh-there, 2017-02-28:
I honestly think all relevant information for super-efficient driving is in "Consumption". The "Braking" screen was specifically added for learning how to do efficient braking: it is very readable in driving conditions and it is giving fast feedback. In reality, anyone interested in "non-friction driving" will master it in a couple of days/weeks and will not have the Braking screen on after that that.

The consumption screen is carefully layed out: from top to bottom it goes from immediate feedback to behavioral changes. I see one change though based on your suggestions that I am considering implementing. There are now three lines depicting SOC in different forms (%, energy, range). This is unneeded information overload. Replacing "Verfügbare Energie" (Available energy) by battery temperature would make sense.

But let me add that in the new Climate screen, there is an IMHO very nice graph re. the temperatures, including battery temperature.

I will leave this issue open until implemented. @fesch do you agree about this proposed change?

wolfijenne, 2017-03-01:
Thinking about this whole thing, I think it has more to do with personal preferences than with what is best. For example, I don't like graphs in such a tool, since the graphs can not be used while driving (since to tiny) but more afterwards for analyzing. For analyzing I like more tables and log files as I am able to generate my own graphs out of it. However, I fully understand that others might love that feature. 
I think my original request more relates to the desire to be able for personal customization of the screens, which would be a different request and probably a whole new project. Therefore I agree with you on your proposal. And yes, battery temperature is very important information.
Thanks for all the effort you spend for this project!

yoh-there, 2017-03-01:
Heads up is always appreciated, and agree with your analysis.

There are no plans right now to make CanZE a "smack widgets on a canvas" application like i.e. Torque light.

The log functions in CanZE are limited but will be improved (hopefully) over time. Right now it can log what it displays.

yoh-there, 2019-01-13:
Closing due to lack of activity. Consumption screen has been slighlty modified since (Delta with reality added, removed absolute energy in battery). I don't see the current developers doing big numbers instead of the graphs.

closed 2019-01-13 by yoh-there