fesch/CanZE #445: Feature Request: Brake Screen

wolfijenne, 2017-02-24:
Display Power instead of Torque. Power=Torque*Speed. I think it would be more relevant. So applying the break pedal with speed=0 should have 0 reading (as there is no waste of energy or wearout).
I know, its more important for people living in the mountains.

yoh-there, 2017-02-25:
Not dismissing your request, it is an honest question: have you noticed the blue bar and/or read this? http://canze.fisch.lu/the-blue-aiming-bar-explained/

I am asking is because I think I understand the underlying motivation of your question and we believe we solved that long ago doing EXACTLY that. Actually, since braking torque is more or less proportional to brake pedal pressure, I feel torque is far closer to the drivers "actions" than power.

Again, I am not being stubborn.

wolfijenne, 2017-02-25:
Thanks for the link. I have not read this before. And yes, this is exactly what I am trying to achieve. Since I am living on a mountain with 500 meters of hight difference, this is actually important to me.
Fortunately, the R90 behaves much better than the Q240 I tried before. I can see much higher regeneration power values than the max. charge/regeneration value indicates. (max charge power was as low as 4.1 kW, but I could use 10 to 25 kW for the whole 10 minutes driving down). I believe there is a bug in the car software, since max charge power should also be better with a higher battery capacity of the 41kWh battery. However, I like that bug.
But therefore, we should not use the max charge power to make calculations for that purpose.

For the power v.s. torque discussion: I found it more valuable to have the mechanical brake power indicated than the torque due to the fact, that at low speeds, the Torque can be high, but the "harm" is not big since there is only a little power involved. On the opposite, at high speeds, a low torque can be much more harm to the brakes since the related power is much higher. (The thickness of the brake lining could also be expressed in kWh of brake capacity).

However, I understand that there can be different view points to that subject.

Just a side note: I use the mechanical brake power also on a Toyota Prius ODB App for some years now for the same purpose.

wolfijenne, 2017-02-25:
I thought a bit more about the blue bar concept. I have to admit it is a cool concept and also adds some dynamic to the screen, so keep it. However, I still like to see the power instead of torque :)
I still don't understand the value of Max ChargePower/Regeneration Power. The value given there is exactly used for charging, but seams to have no effect on regeneration. I'll make some more tests.

yoh-there, 2017-02-26:
It has total effect in regeneration. Try regeneration on a full battery :-). Another driver, also living on a hill/mountain did some testing. See this http://canze.fisch.lu/max-battery-charge-and-max-battery-regeneration/ (also read the comments)

wolfijenne, 2017-02-26:
Thanks for the info. Last fall I had a Q240 for a week. There I was not able to use much Regeneration at 80%SOC. Now with my new R90, I was able to use full regeneration for the whole way down starting at 90%SOC. Calculated, that gives a constant 7kW for 10Minutes in average.
It was also what I expected, since the battery capacity doubled.
For that reason I was surprised to see a very low value for the Max Battery Charge in CanZE and having much higher RegenPower over a significant period of time. (Which I totally love)
The value of Max Regen Power actually changes depending on the current currently applied to the Battery which I found also quite surprising. I'll post these in the link you mentioned above.

yoh-there, 2017-02-26:
Maybe #447  is biting you? iOS?

wolfijenne, 2017-02-26:
Ah no... I use Android

yoh-there, 2017-02-28:
I have thought a bit more about the torque versus power and actually did a few more test drives with a keen eye. I disagree on switching to power and let me explain why.

On a particular braking run, max power is a given near-constant, dictated by component sizing, SOC and temperature. I think we can agree on that. I think we can also agree on the goal being to minimize friction braking / maximize regeneration. It then alls comes down on where to aim with the braking pedal.
On an ideal braking run, power would remain constant and close to the maximum, constant regen power (assuming distance is available of course). To apply constant power braking in a brakeing run requires more torque as the speed goes down. The Torque bar gives this feed-forward indication. It creeps up as speed goes down, indicating to the driver he can/should brake harder. In a display-power scenario, the aim point would be fixed and as speed goes down, so would power, ending up in a more feed-back scenario.

The difference is, I admit, superficial, but "controlling" the drivers brake foot through torque is a more linear and natural mechanism.

Now there is probably an effect too that driving down a long slope in hilly terrain is not the same as coming from to a stop from 120 km/h in my "flatland" ;-) I appreciate that difference.

Finally, I am sure what one is used to plays a role too.

wolfijenne, 2017-03-01:
For the aiming game (basically the blue bar), I agree with you, it does not matter if power or torque is used and therefore we should not spend effort to change that. Actually I don't need that game since I learned with my Prius to brake and it is quite simple: The less pressure I apply on the brake pedal, the better it is for regeneration, consumption, wear off the brakes. :)
I was more into having a numeric value of the friction brake power. A torque would not help here since the torque is 100% when speed is zero but has no effect.

In my daily brake game it is only relevant for the downhill portion, since there the SOC, Battery Temperature are possible limiting factors. Also the ABS could be such a a factor. In my Prius, when applying the brake to about 10-20kW of regeneration while driving over an "obstacle" like an metal cover on the road can cause the ABS to kick in and stopp regeneration almost completely. It will not recover until the brake pedal is released for some seconds. I don't know yet how the ZOE is behaving in that situation. So for such reasons I am happy to see just the power value of the friction brake power.

yoh-there, 2017-03-01:
I've never had the ABS kick in on a bump and stop regeneration, though a very hard or a very quick appliance of the brake pedal will do that as to have full control over all 4 wheels as opposed to the differentially coupled front wheels when regenerating. From own experience I can testify it works very well, ahem.

Worst that happens if you take a speed bump very aggressively on cruise control it will kick off.

As for energy conservation, my strategy is: preferably coast (in N or aiming at 0 kW, same effect), which means extreme anticipation, next regenerate (not exceeding blue bar), next "do as needed".

Closing this issue.

closed 2017-03-01 by yoh-there