fesch/CanZE #446: Feature Request: Fixed Colors for Battery Heatmap

Juergen63, 2017-02-26:
Is it possible to have fixed scale for the battery temperature? That's mean 10° have always the same color. Maybe there is an option to change to color which however want.
This could be easier to compare screenshots of the heatmap.

Thanks Juergen

yoh-there, 2017-02-26:
The goal of the heatmap is not to compare over time, although one could very easily use screenshots as the absolute numbers are given. The colors are now relative to the mean value, to clearly indicate cells that over- or under-heat, which would indicate a real problem. That (main) function would be lost changing the color scheme, while the information the OP seeks is readily available. Personally I would vote against this request. Maybe I missing a point. Others can chip in of course.

Juergen63, 2017-02-26:
Actually I test a external battery heater under the battery. I lock how is the distribution of the temperature for different cells over time. With all screenshots I build a small movie and have a time-lapse. So it's easier to find hotspots.

yoh-there, 2017-02-28:
I am assuming this is a one-off issue then. I suggest to take the absolute values to excel, either by hand or by using the logging function. Another option is to use a CANbus logger, such as the Freematics One. I am closing this issue

closed 2017-02-28 by yoh-there