fesch/CanZE #447: Available Charging Power changes between 2 values

HarmOtten, 2017-02-26:
iOS. v1.0.15.
The Available Charging Power on the "Charging (technical data)" screen changes between 2 values during one charge session. One shows 1 phase with a low power value, and the second shows 3 phases with a higher power value.
I think this is incorrect.

yoh-there, 2017-02-26:
From a corresponding blog entry in the Dutch forum I take it that this is a specific issue on the iOS version. So I am marking this as such. If this is incorrect, please let us know in this thread.

HarmOtten, 2017-02-26:
Yes, this is only an issue with the iOS version. Please mark as such.

FredLeudon, 2017-02-26:
Thanks, i will have a look. 

Frédéric RICHARD

FredLeudon, 2017-02-28:
Will be solved in version 1.0.18
Added balancing module datas to technical charging activity as in Android version.

closed 2017-02-28 by FredLeudon