fesch/CanZE #449: Bug in Logging

wolfijenne, 2017-03-03:
There seems to be a bug somewhere in the logging. After some time I have seen Timeouts on the screen and also screen values did not move anymore. I could fix this problem by starting the Bluetoothadapter again. However, the data logfile did not notice this and has forever recorded the last measured values before the problem occured, although values on the screen were measured (see attached file). Unfortunately it will be hard to reproduce.
Last measured value was written @2017-03-02-19-55-01. Car was charging until approx. 20:30, after that driving. 
Enclosed also a screenshot from the charging where a gap can be seen at 19:55.
[data-2017-03-02-18-14-52.txt](https://github.com/fesch/CanZE/files/816245/data-2017-03-02-18-14-52.txt)
[image]

fesch, 2017-03-03:
If the dongle falls into an unidentified mode or blocks, it has to be restarted. This is a know issue of the dongle and we are trying to find a way to revive it from the side of the application. Sometimes it lasts for hours, sometimes this happens already after a few minutes.

The mentioned logger (I did not write this!!) issue comes from the fact, the this class writes every 5 seconds a value to the log file, despite of the fact if there has been a new one that arrived. Chris should fix this...

I am working on some new logging and user configurable logging facility that will replace all this.

wolfijenne, 2017-03-03:
Thanks, I'll happily wait for the new logging feature then :)

closed 2019-01-06 by yoh-there