fesch/CanZE #450: Help wanted, CanZE for iOS translations

FredLeudon, 2017-03-04:
Hi all,

I need help to translate iOS CanZE.

See attached files. 

You can send translated files to ioscanze@icloud.com or re-attach to this issue.

Thanks

[extract_ES.txt](https://github.com/fesch/CanZE/files/818584/extract_ES.txt)
[extract_FR.txt](https://github.com/fesch/CanZE/files/818583/extract_FR.txt)
[extract_GB.txt](https://github.com/fesch/CanZE/files/818582/extract_GB.txt)
[extract_NL.txt](https://github.com/fesch/CanZE/files/818581/extract_NL.txt)
[extract_SL.txt](https://github.com/fesch/CanZE/files/818585/extract_SL.txt)
[extract_DE.txt](https://github.com/fesch/CanZE/files/818580/extract_DE.txt)

yoh-there, 2017-03-04:
My kind suggestion would be to write a little program to process the XML
files to TXT or, if that is not possible due to lacking info (keys etc) to
use a common format. Sound like a silly job to go over the translations
twice no?

sordoneF, 2017-03-05:
I can do the Dutch (and maybe German) translations but it is bit unclear which texts to alter. Only the texte:5 fields ? What about Texte:35 field ? Some msg fields also have "messages". I Suppose the champ: fields must be left as is.

FredLeudon, 2017-03-05:
I will try first to match CanZE for Android stings with CanZE for iOS

After all, i'll ask for CanZE for iOS specific translations.

FredLeudon, 2017-03-06:
I close this issue for the moment. 
I worked with xml files on repository to translate strings in my app. 

I will build my own xml specific keys to iOS and I'll reopen this issue when all common strings will be translated.

closed 2017-03-06 by FredLeudon