fesch/CanZE #452: Charts: Color of description the same as color of line

HarmOtten, 2017-03-07:
CanZE Android version.
Is it possible to make the color of the description of a line in the chart, the same as the line color?
This would make it much easier to read the charts.
This applies to the Climate screen, and Charging graphs screen.

fesch, 2017-03-07:
Should be. The problem is, that actually the text is being written within a single line of code. I'll check this ...

fesch, 2017-03-07:
... and is configured as a single line in the views config :(

yoh-there, 2017-03-07:
Comma separated, like the SIDs? Or is that too complex to position nicely, as they become separate entities?

fesch, 2017-03-07:
Must use that ... because the slash is no option --> km/h  :(

fesch, 2017-03-07:
Fixed in dev branche

closed 2017-03-07 by fesch