fesch/CanZE #453: Charge screen: Show temperature range

HarmOtten, 2017-03-07:
Is it possible on the basic charge screen to show the temperature range (instead of only the average temp)?
So for instance: 15-18 °C
I am asking this, because in many cases the temperature of the coldest battery module determines the charge speed (and in hot situations probably the warmest module).

In my opinion it would be sufficient to show the same on the Detailed charge screen, because a separate Temperature heatmap screen that shows all battery modules is also available.

yoh-there, 2017-03-07:
AFAIK the LBC has minimum, maximum and average fields, so this should be
possible

yoh-there, 2017-11-11:
Superseded by #471 which I might build as stated.

closed 2017-11-11 by yoh-there