fesch/CanZE #457: Timeout on [atma] (Android)

Albino-Zoe, 2017-03-09:
Hi,

this morning I had this message all the time, no screen responded.
Reboot ELM, CanZE, Tablet unsuccessfully.
Since I still worked with V1.22, I installed V1.24.
Same behavior.
I removed CanZE, installed V1.24 again.
Since then no more reports of this kind.

What does this message "Timeout on [atma]" mean?
I only found it in #24 (ELM performance).

Logging is available.

-------------------------------------------------------------------------------------------------------------------
Zoe Intens R240 (25.08.2016) 25.000km/Jahr CanZE (Android) (V1.24) mit KW902 ELM327 (27.01.2017)

yoh-there, 2017-03-09:
ATMA is a command to the dongle to monitor for a "free" frame (a frame that is just transmitted by the car without anyone asking for it. It is hard to get the timing right, as the dongle can be overwhelmed when we're dealing with a frame that is transmitted very rapidly.

Your mileage may vary. I suspect the Bluetooth stack in Android (depending on chipset used) is sometimes flakey. FWIW: I need to restart things too quite regularly.

fesch, 2017-03-09:
Me too. It's not stable at that point. Works mostly, but I also already days I needed to restart quite often before it worked :/

Albino-Zoe, 2019-02-07:
O.K. in glad expectation of the new dongle I leave the topic closed.

yoh-there, 2019-02-17:
The dongle is a DIY project (unfortunately), but if you have some basic soldering skills it is not hard to do. I am now running this thing for something like a month without firmware updates and I can just say it is an absolute killer compared to the commercial ones, and roughly the same price in parts. Bob put in some decent documentation

https://gitlab.com/jeroenmeijer/cansee/tree/development