fesch/CanZE #458: fonctionnalité rendement

langloismichel, 2017-03-10:
bonjour,
je souhaiterais pouvoir lire sur l'app une estimation du rendement de charge effectuée par le chargeur caméléon de la Zoé, notamment quand il est branché à domicile sur le cable flexichargeur
merci

fesch, 2017-03-10:
Comme la voiture n'a pas d'information relatives à la consommation réelle, cette information n'est pas intégrable dans CanZE. Dans ce cas il faudra s'orienter vers une station openEVSE.

fesch, 2017-03-10:
@Jeroen: The question is about displaying the efficiency of the charger inside CanZE, but I think this is not possible as the car does not know what is really being consumed.

yoh-there, 2017-03-10:
We can dig up DC power (Charging screen) and AC power (BCB screen, I hope it works on your end), so we could display DC on the BC screen too, and/or a ration between the two

fesch, 2017-03-10:
But will that give the efficiency of the charger?

yoh-there, 2017-03-10:
Well, isn't that by definition the efficiency of the charger? 10kW AC goes in, 8kW goes to the battery (assuming no serious consumers during charging)? Or am I misunderstanding the question?

fesch, 2017-03-10:
@langloismichel : Est-ce que ce va avec l'anglais ou devrais traduire?

langloismichel, 2017-03-13:
je préfère le français, mais bon je crois comprendre

><https://github.com/langloismichel>@langloismichel 
>: Est-ce que ce va avec l'anglais ou devrais 
>traduire?
>

yoh-there, 2017-03-17:
I am not sure if the addition I proposed is answering the OP's question. That might have been unclear when I said "or am I misunderstanding the question".

If there is lack of feedback I will close this issue soon.

yoh-there, 2017-06-08:
Closed for lack of feedback from the OP

closed 2017-06-08 by yoh-there