fesch/CanZE #459: iOS Trnaslations

FredLeudon, 2017-04-04:
Hi all

I worked hard to add Android translations to iOS CanZE.

Could somebody translate specific translation attached below ?

Thanks a lot.

[translations.txt](https://github.com/fesch/CanZE/files/892122/translations.txt)

FredLeudon, 2017-04-04:
It is also necessary to translate this screen

[image]

yoh-there, 2017-04-04:
Dutch:
Language -> Taal
WiFi settings -> WiFi instellingen
Automatic conexion (watch out, english spelling error: connection) ->
automatisch verbinden
Car -> Auto
Tyres pressure sensors -> Bandenspanning sensoren
other (capital O) -> Overige instellingen
Miles mode -> Mijlen modus
Field loggin (Field logging) -> Veld logs
Safe driving mode -> Veilig rijden modus

wolfijenne, 2017-04-19:
I tried myself on the german translations, and where not done yet, on the english.
Whenever I am not shure, I put in a comment: (not sure on this one) 

[translations_de.txt](https://github.com/fesch/CanZE/files/938899/translations_de.txt)

**And the settings screen:**
German:
Language -> Sprache
WiFi settings -> WiFi Einstellungen
Automatic conexion  ->
Automatisch verbinden
Car -> Auto
Tyres pressure sensors -> Reifendrucksensoren
other (capital O) -> Andere
Miles mode -> Meilen Modus
Field loggin (Field logging) -> Feld Loggen
Safe driving mode -> Sicheres Fahren

hope it helps

manuelrocha88, 2017-04-25:
I've translated to Portuguese (Portugal Version) only the cDevice messages. It seems the messages are the same for the c E L M 3 2 7

If some have missed please ask and I translate it for you.

[translations_pt.txt](https://github.com/fesch/CanZE/files/955450/translations_pt.txt)

And the settings screen:
Portuguese:
Language -> Idioma
WiFi settings -> Definições WiFi
Automatic conexion -> Ligação automática
Car -> Carro
Tyres pressure sensors -> Sensor de pressão dos pneus
other (capital O) -> Outros
Miles mode -> Modo Milhas
Field loggin (Field logging) -> Histórico
Safe driving mode -> Modo condução segura

mrubioroy, 2018-02-21:
Spanish and Catalan translations.

I've also made some corrections to the English translation. Please note that in English, ponctuation marks are not separated form the text: "The following:" and not "The following :" or "Hell!" but not "Hell !". However, this separation is common in French.
[translations.txt](https://github.com/fesch/CanZE/files/1744074/translations.txt)

yoh-there, 2019-01-13:
Can I close this issue @FredLeudon ?

yoh-there, 2019-01-28:
Closed due to lack of activity

closed 2019-01-28 by yoh-there