fesch/CanZE #460: Feature Request: Remote Charge Tracking

wolfijenne, 2017-04-07:
When charging, the car does not report its charging state to the Renault App. So it is not possible to monitor the SOC from a distance since the App also does not allow many state requests in a short period of time.
The idea would be to allow uploading certain parameters from the App to a Server. From there it could be examined via a Webinterface or a different App. The car owner would need to have a phone with a SIM card connected to the dongle for that feature.
Data transmitted could be SOC, Battery Temp, DC Power and Battery Voltage for example.
In principle, it could also include Car speed and GPS Data from the phone allowing for tracking.

For the server side programming and providing, I could imagine to volunteer. Let me know what you think.

fesch, 2017-04-07:
Since the start of the CanZE project, all thoughts regarding "tracking" or "uploading data to some server" have been discarded for obvious security and privacy concerns. Per personal opinion is that CanZE should not deal with something like this.

If I would like to remotely ready data from my car while charging, I would take some other options like using an MQTT enabled device that connects to the CAN bus and transmits data to my broker. I think we should leave this to each interested person itself. 

But maybe we could implement some "howto's" for guiding ... ?

wolfijenne, 2017-04-08:
Thanks for the answer. Although I don't really see the point of Security and Privacy when monitoring charge data. Let me know if you have some documentation about the howto. Thanks.

yoh-there, 2019-01-05:
I am closing this issues. If we implement OTA MQTT, it's covered. If not, well, then not

closed 2019-01-05 by yoh-there