fesch/CanZE #462: wichtige Batteriedaten werden nicht übertragen, Rest schon

goldi1510, 2017-04-14:
Kangoo ZE Maxi Bj 2012   km 72.000
Maxiscan 902
in der App werden zwar viele Werte angezeigt, z.B. die Zellspannungen, der SOH 89%, etc. aber:
die unter "Batteriewerte" angezeigten ersten fünf wie Positionen nutzbarer Ladestatus, verfügbare Energie, Energie bis voll geladen etc. werden nicht angezeigt. Egal ob App auf Tablet oder auf Galaxy S7.
Da die Reichweite des Kangoo momentan nur noch ca. 100 km bis absolut Stillstand / geschätzte 16,5 KWh statt 22 KWh hat, möchte ich genau diese Werte messen. 
Zum Eingrenzen meines Problems folgende Fragen:
Kann es auch am Fahrzeug selbst liegen, daß ein Teil der Elektronik defekt ist, oder sieht es eher nach Problem mit dem Adapter aus?
Oder liest die App bei einem der ersten Kangoo ZE aus 2012 evtl. nicht alles aus?

yoh-there, 2017-06-08:
Sorry for the very late reply. Unfortunately we do not have a Kangoo to test code changes, so it is very hard to determine what works and what doesn't. I can try to apply some fixes, but it would be a slow "hit-and miss" project and you would need to do the testing.

Can you send a screen shot of the offending screen?

yoh-there, 2017-11-11:
Close due to lack of requested feedback

closed 2017-11-11 by yoh-there