fesch/CanZE #464: enhacement

FrancescZOE, 2017-05-03:
I am willing to have a "session" or kilometer A or B similar funtion. Where one could push and say to CanZE. hey please beging accounting kwh.. and end when i push again. So one could track Total KWH in partial drivings. The ZOE r-link is so bad, just in every start stop. Not in a full trip.   IT could have "simple" user viw like km done... kwh used... and % batery used and kwh/100km... then another one "more profesional" with just a "-" screen showing the comsumtiops of all... the changes i mean. Baterry temperature increase or decreace.. kwh.. increase.. voltage cell "variations"... tempetarure grow of each cell... and so one :-)

 ##

fesch, 2017-05-03:
That's a nice idea ... BUT ...

Generally the BT link is not stable enough to allow your device to be connected to the car all the time. If the link breaks, accounting would be broken too. As this will happen quite often, I think that investing time in some functions like this will not be worth the result.

But if you want to do that, just go ahead. :-)

FrancescZOE, 2017-05-03:
ummm.. so there is no way to "save" results on a log and use them later?¿ lets say... "get a capture of values" save on .txt, stop BT .. when the trip finishes...  start again and re-capture. Compare and "vuala" :-).

P.D i have notices that when BT is on... my car does not "breaks" really well on some surfaces. So it makes dangerous to have in ON during criving. I think you have wrote that on notices to the "license" to use can ZE .

fesch, 2017-05-03:
Please go to the settings and enable the logging.

CanZE does only read data, not set data. Thus the problem with breaking seams strange. Did you do some other manipulations?

FrancescZOE, 2017-05-03:
Hello, i have from begining... at 2016,  SD card logging-> Debug+Fields   and also "Log to sdcard1".
I think is a thing of ODBC2 bus... and Renault. when there is a BT conneted... data is "shared" on the same bus. So using bandwith... and extrangely affectig ESP+ABS... When switch OFF BT... reader... no problem at last so notable.

fesch, 2017-05-03:
@yoh-there : do you have an explanation fir this? I don't ... and never heard from any other user such things.

@FrancescZOE : What dongle do you have?

yoh-there, 2017-05-03:
This is all "my opinion" of course...... 
- No, I never saw any of this;
- Possibility one: disturbance of the CANbus. Possible, for instance if the dongle has an end-of-bus resistance pair, which it should not have. What is going against that hypothesis is: why on earth would it affect the brake system only? Basically, it would make more sense if the entire car would behave erratically. If the car behaves normal with CanZE off (and background not enabled) I think we can safely dismiss this theory.
- Possibility two: CanZE + the dongle interferes on a message level with the ESC and/or UBP. The only commands that do this are, as far as I know, the DTC commands, which of course should never ever be used in a rolling car. Apart from DTC's, we never target the ESC or UBP with diagnostic commands (760 and 7bc).

So, indeed, I would be very interested in which dongle you use, and which screens give you trouble.

Finally, the car has indeed little recording posibilities, though there are some totals kept, which could be used before and after a trip. I have no appetite to build it, but would be happy to share the commands we have knowledge of.

Most of all: be safe.

yoh-there, 2017-06-08:
Closing for lack of feedback fro OP

closed 2017-06-08 by yoh-there