fesch/CanZE #465: XTOOL iOBD2 mini

manuelrocha88, 2017-05-03:
I have an iOBD2 mini from xtool company and I've connected it to my ZOE.

From iOS CanZE APP I can see the equipment but when I try to connect it stays at initialize pop-up and nothing happens.

Is there any way that I can PULL the iOS App Code and try to find what's happening and make this app compatible with this OBD?

FredLeudon, 2017-05-04:
Hi, instead of CanZE for android, CanZE for iOS is not developed directly with x-code and sources are not available.

Using iOS LightBlue app (easily findable on Apple Store), Can You send me services and caracteristics address ?

As I explained in other posts, I can't buy each dongle available to test it.

If you want, send a mail to ioscanze@icloud.com and i will reply with address you can send Your dongle. Once I get it, i will update code to try to make it work with CanZE for iOS and i will send it back to you. That all i can do to help.

manuelrocha88, 2017-05-04:
Here you have the pictures from the app you ask me to run.

And here is a picture of the top of the device.

Will this help?

manuelrocha88, 2017-05-04:
Here you have the pictures from the app you ask me to run.
[image]
[image]
[image]
[image]

And here is a picture of the top of the device.
[image]

Will this help?

FredLeudon, 2017-05-05:
I will try to do something to help.

But I have a big memory issue problem to solve first ;-)

manuelrocha88, 2017-05-05:
@FredLeudon that's why you should publish the code to the community even if it is not the native code.

Probably you have people here that understand that and have already the same memory issue you are getting.

FredLeudon, 2017-05-05:
Not sure, I'm using Windev Mobile from PcSoft. 

If someone is interested, I can give access to source via GDS.

FredLeudon, 2017-05-05:
I have to work tomorrow an I hope I will have time to search to solve this issue. 

After connecting to Bluetooth LE and interrogating some frames, memory used by app is growing up extremely fast to an <ou of memory> error. There is perhaps an issue in my code and I have to check it before investigating further.

That is difficult because I have to run app on my iPhone in debug mode from my iMac. But car is a little bit too far away and Bluetooth connection isn't stable.

manuelrocha88, 2017-05-06:
You are using the express or commercial edition?

Maybe I can help you solving the issue.

Or, if you want I can help you migrating this to swift language, which is native, and with this we can get probably more people involved.

FredLeudon, 2017-05-06:
I'm using commercial version.

I use pcsoft products every days from most than 15 years ago and I'm not sure to want (have time) to learn a new programming language ;-)

FredLeudon, 2017-05-16:
iOBD2 mini from xtools doesn't seem to be an ELM compliant dongle.

Any help on how to 'talk' is welcome.

manuelrocha88, 2017-05-16:
Did not understand. Do you want to talk directly with me or how the iOBD2 mini talk?

FredLeudon, 2017-05-16:
I need help to find commands that have to be send to iobd2 mini to establish connection, querying ECUs, etc.

manuelrocha88, 2017-05-16:
How can I give you that? Do you have any software for that?

FredLeudon, 2017-05-16:
I installed iOBD2 app but it doesn't help.

I googled but I didn't find anything..

yoh-there, 2019-01-13:
Closing this issue due to lack of activity

closed 2019-01-13 by yoh-there