fesch/CanZE #471: Show how well the battery is balanced

HarmOtten, 2017-11-01:
Showing all the voltages of every cell-pair is nice, but what you also want to know is the difference between the highest and lowest voltage. This is a nice indicator of how the battery is balanced.
See example in the Tesla-App: [https://youtu.be/F3tF0i98MpI?t=45]

yoh-there, 2017-11-01:
That is why that screen is color coded. No need to hunt for numbers and interpret them. The more the colors difference, the more there is a problem. Same with temp heatmap.
Note that the colors represent "deviation from the average value".

HarmOtten, 2017-11-01:
The colors are indeed very nice to find the lowest or highest value, and gives you a good indication if something is wrong or not. Nevertheless I think the difference between highest and lowest voltage would be very usefull. You could for instance show it on the battery-screen, or maybe on the detailed charge-screen. At the end of the charge-process (at 99%), it will ballance the battery. Would be nice to see the number to down at that point.

yoh-there, 2017-11-01:
I honestly think it is not very useful. You've kinda failed to make your case ;-) Of course it can be shown on a screen, but say it would display "67mV" for a while? Then what? Maybe @fesch  thinks differently.

But thinking about gave me a maybe better (as in more useful) idea.

New screen "battery balancing"
6 blocks representing SOC 95%, .... 99%, 100%.
If the car reaches said percentage (so the event is last value = SOC%-1, current is SOC%) said block would be populated with
- current time & date
- min and max cell voltage (there you go ;-) )
- DC power level
- 12 balancing switching hex numbers (same as in tech charging)
- 12 module temperatures (same as in tech charging)

If suspecting a battery problem, one could let the phone sit in the car and see what happens.

fesch, 2017-11-01:
I don't see why we could not display that value. After all, it's simple to get, but I think I won't need getting updated to fast either, so this will no hurt.

yoh-there, 2017-11-02:
For (stubborn!!) me, it's not a matter of can; it is indeed easy to do. I am trying to wrap my head around the perceived usefulness. My suggestion above is considerably more work  but I can see that actually helping users.

fesch, 2017-11-02:
I think, that a single value can only indicate a single thing, In this case: Everything is alright or there is a problem. This indication only needs a single value. 

So what I personally see as advantage here is, that this does not take too much space on a screen where other information is already displayed. That's all ...

yoh-there, 2017-11-02:
But now we're back to the color coding (IMHO). I know I may sound argumentative, sorry, I just want to do this right if we do it.
The other thing is that I suspect (not 100% sure) that the significance of the low-high gap might only show at certain (very high) SOH levels, then canceled out by the balancer.
Again, I have no problem by adding the field (it's a scrollable screen anyway); it's more the interpretation of it.

closed 2019-01-05 by yoh-there