fesch/CanZE #473: Ganze on android with Q90 miscellaneous issues

sts21, 2017-11-03:
Real soc is shown as 0.0
Max Battery recharge is shown as 0.0
Energy to full is shown as -
Battery temperature values are shown like:
9/10/9
-39/NaN/Man
63/96/95
103/72/79

Mileage battery range is fix 256km no Change
Charging prediction is not shown 

Firmware
LBC soft and version is shown as 0000
LBC2 Version is shown as 0000

yoh-there, 2017-11-04:
Sounds like a problematic dongle, but to be sure we need some more info:
- what kind of dongle?
- what CanZE type (Android or iOs)

sts21, 2017-11-07:
I am using an no Name china Elm327 Clone which was Working fine on my R240 on the same Android System! 
But to be shure its Not the dongle i ordered the recommended KONNWEI KW902 on Amazon. So i will try this first and let you know about my test results!

Greetings sts21

sts21, 2017-12-02:
Today i could Test the new Konnwei dongle und got the same problems.
I although tested both dongle on an R240 again in wich both where working fine without any problems.
While testing i discovered that, when i close the car an wait until all Systems went in standby and i than open the car and check the values in CanZE moste values are shown correctly!
After waking the car up i get all the Problems mentioned in my first Post.

Although i found out that I am not the only one with an Q90 that has these Problems:
https://www.goingelectric.de/forum/renault-zoe-zubehoer/hacking-into-zoes-canbus-t12199-370.html#p640213
https://www.goingelectric.de/forum/renault-zoe-probleme/zoe-q90-keine-leistung-mehr-t26844-20.html#p609510

If you need more information or have any suggestions what else i can try feel free to ask!

yoh-there, 2017-12-02:
Thank you for posting that. It seems a more serious problem with at least some fields. I must conclude the LBC of the new battery is substantially different. Unfortunately, we do not have a Q90 at our disposal. I will ask around if I can get access to one.

fesch, 2017-12-02:
Another user contacted me with the same issue on his Q90, so these problems seam to be confirmed.

yoh-there, 2017-12-02:
Mate just checked (Q210 with upgraded battery). No problem to show heatmap and temperature map. Now I am confused. We need to find a car closeby either @fesch or me I guess to get to this, ar do a quick diagnostics activity in the next release, maybe showing the raw response from the dongle for a few of these commands? If the LBS is really different we're kind of screwed, as I do not have documentation and do not have access to do reverse engineering. Hmmm.

yoh-there, 2017-12-02:
To anyone stopping by here on this issue and owning a Q90:
1) are your heatmaps and voltage maps OK? (so we know if this is a generic problem or not);
2) do you also get all 000's at the firmware versions for the LBC? If not, can you please share a screen shot of the LBC firmware and state if the heatmap and vltage map work OK or not?

R240Pe, 2017-12-15:
2 weeks ago - after waiting for 6 month - i got my new - German! - Q90 ZE40. My first test with the Konwei Dongle showed wrong values.  Because it first does not machnically fit , so i had to do some mechanical adjustments. 
After it fits, it worked fine and my temperatur an voltage-values seemed to correct (first row screenshots).
Yesterday I had to charge up my ZOE starting fom SOC 1 % and I wanted to control the charging by CanZE with my Sony Xperia Z3.
But this time CanZE showed very weird data. There was no colour for the temperature and the voltage varied from N.a.N to 65,xxx Volts. Waiting for minutes the values of the cells varied but the values did not beecome correct. Wrong values sometimes became - perhaps - correct, whereas possibly correct values showed reliable wrong again.  Later after several new starts of both Dongle and CanZE I took the screenshots in the second row.
I also tried a MEDION Tablet; it showed different values than on the Xperia but also wrong values and the same behavior.
This time there were also values missing, like already discribed above, for example the "Laden -> Max Batterie Ladung" or "Laden -> Echter Ladestatus"- they showed "0". Last time the values were correct - i controlled it by iCarSoft. 

Because I´m logged in here today for the first time, I did know and not do the test with the "sleepy" ZOE like described above.
As soon as possible I will do a test with the dongle when ZOE fell asleep. I´ll post the results here.

[image]

R240Pe, 2017-12-15:
I´ve just finished the "sleepy ZOE-Test". At first I disconnected the Konnwei Dongle and waited till ZOE was totally asleep. Then I connected the Dongle again. I pushed the power on button on the dongle and startet the app. Bluetooth connection was build up and connection worked corretly, but no data were received. I tried this tree times - allways the same - no values. If Zoe is asleep, there is power on the Can-Bus but ZOE does not send any data.
Then I disconnected the Dongle and stopped the app. I only opened the door of ZOE, connected the dongle and startet the app. Connection was built up and ....!!!! all gathered Informations were ok. The values seem to be correct and as far as I see completed (not yet controlled with iCarSoft).
I also repeated this tree times - all ok.
The screenshots are the same as on my last post - upper picture-row - all vallues seem to be ok.

Next with activated connection between dongle and android-handy and correct values on the heat- and voltage-map  I pushed the "START"-button on the car dashboard. Immediately values and colours got crazy and showed up like my recent test(-values lower picture-row). At the same time some of the values - as described above - disappear and "0" or nothing is shown.
So I think there is something  as a result of pressing the start button, which makes the values "crazy".
Perhaps this information gives a little help for the solution of the problem.
Concerning that behavior, I do not think first, that it is a dongle problem
Some others should do the same test, to see whether their ZOE´s get the same results as I on my German Q90 (11/2017).
Greetings R240Pe - now Q90Pe

fesch, 2017-12-15:
Hi

I think this is a dongle problem, because of this: When the car is awake but the motor is not powered on, there is a certain data flow on the can bus, which increases as soon as the start button is being pressed. At that time the bus holds more than 1600 messages/second.

I could easily image that some of the dongle out there are not fast enough to catch up with that many messages. At the very beginning of the CanZE development, I used an Arduino Leonardo and ended up with an Arduino Due because the first one was not fast enough to intercept/read all messages.

R240Pe, 2017-12-15:
Thanks for this information - i didn´t know this exactly.
Another thing I recognized in my screenshots an on most posted screenshots - if there are wrong values, it always affects the same data fields. Mostly the same data fields have NaN or much too high values. Some values are always correct! Is there a rule how this values are sent and a "timeout" causes the same precedure of creation of the crazy values?

R240Pe, 2017-12-15:
Additional question:
on R210 and R240 - as some users reported - dongles work fine - the same dongles don´t work on R90 / Q90!
If it´s the speed of the dongle:
Is data flow on Q90 can bus faster?  or more values?

yoh-there, 2017-12-15:
Databus speed is the same on all models, 500 kbps, but for the R240 and the Q210 the bus loading when started is roughly 30%.

I see no reason why it should be higher on the 90 models, but hey, something doesn't work.

Maybe the new design 902 dongle can't handle the data flow very well; I see it has problems with my Q210 too (though there it simply misses a lot of data, it doesn't seem to screw it up). Hmmmm.

R240Pe, 2017-12-19:
Today I tried my Konnwei dongle on my old R240. There everything worked fine, even when pressing the Start-button. The Voltage and heatmap did not change after the button was pressed and no NaN or empy values appeared.
So I think, my dongle must be ok. 
I also think, there must be any difference how it works on the R210/240 and the R/Q90 when the Start-button is pressed. 
Is there anyone, who tested the CanZE with an arduino on R90/Q90? Are there the same results?

R240Pe, 2017-12-20:
Last night I charged my ZOE. After opening the door, activating the dongle and CanZE, all values were shown and seemed to be correct. Then I plugged in to my NRG-Kick, charging started,  I had a look at CanZE - the shown or not shown values looked like I pressed the Start-button. While the battery was frozen (-1 °C) it took nearly 5 hours to Charge the battery. Waiting some time and waking up the ZOE by opening the door, activating dongle and CanZE I got normal values again.
The behavior by pressing the Start-button or charging the ZOE is the same - wrong or missing values.
IF it´s useful, I am able to load up some screenshots.

yoh-there, 2017-12-20:
Thank you for the detailed investigation! Can you help us a bit more please? I think you have stated you use a KONNWEI 902 dongle. Now there exist 2 versions (actually 3, one is buggy). I hope you are able to open it up. Have a look here please. 

http://canze.fisch.lu/dongle-trouble-3/

Does your dongle looks like "Latest" or like "Old"? And if it is latest, can you see the value of the resistor  just above the 8 pin chip in the middle? It should read either "121" (120 ohms, which is OK) or "200" (20 ohms, which is NOT OK).

If, and only if, your dongle is "Latest, 200", please follow the instructions here http://canze.fisch.lu/dongle-trouble-4-solution-sort-of/ and please report the result.

Don't foloow said instructions if it's an "Old" one, or a "Latest", with a "121" resistor, but please report what you have. Thanks again.

R240Pe, 2017-12-22:
This is the dongle i ordered:
[https://www.amazon.de/Konnwei-ELM327-Bluetooth-CAN-BUS-Scanner/dp/B075MY2LVS/ref=sr_1_10?ie=UTF8&qid=1507570238&sr=8-10&keywords=Konnwei+KW902](url)
It is SerialNr. X000038KHV
Description on package is:
**Support
J1850 VPW J1850 PWM
IS09141-2
KWP2000
CAN**
The dongle looks like "Latest"
[image]

and the resistor reads about 119,3 to 119,7 ohms
[image]
The measured pins can be seen in the photo.
I think, it is ok - nothing to do.
If you need further help - and I can do it, please tell me.

yoh-there, 2017-12-22:
Yes it's OK, thanks, it's a "latest", with the proper resistor fitted. I actually have no clue what's going on to be honest. Unfortunately I also don't have a Q90 model at my disposal to test and fiddle with. I will drop a line in the Dutch forum to see if there is somebody closeby  I can swap with for a few hours maybe.

R240Pe, 2018-01-12:
Today I found this video on YouTube. [https://youtu.be/GSp0bjyHySI](url)
An english ZOE Q90 charging at a 43 kW quick Charger. At about 5:00 CanZE on Samsung Handy with Bluetooth (dongle is displayed at 8:06) is shown and surprisingly on "Technical Charging" correct values are reported while charging. Also on the "Main Charging" screen the shown values seem to be correct. While starting charging with my Q90 I never had any useful information in the "Max battery charge"-field, neither the value was displayed with a red background.
At 8:13 a correct heatmap of the ZOE can be seen while the EV is charging; at 8:33 a correct voltage heatmap can be seen.
So what´s the reason that things are fine in the video? What´s the difference? I think Renault uses the same Can-software on both the english and the german ZOE. Is it the dongle "MaxiScan"?

There is another thing I recognized. Waking up the ZOE values are correct. Pushing the Start-button while pressing the brakes values stay correct. Values change to incorrect when releasing the brake pedal.

yoh-there, 2018-01-12:
Dang, just came home from a '90 owner and we could not replicate the erratic bahavior, but didn't try Start / brake sequences. Oh well ;-) 

GOOD INFO, thank you. As soon as I have access to another model with a ZE40 battery, I will try this way and see what we can figure out.

kresnikf, 2018-02-27:
I have the same problem on ZOE Q90 ZE40 2017 bacause On my Q90 dongle showed wrong values. I tried my Konnwei dongle on friend R240 work 100%. 
Any solution ???

R240Pe, 2018-03-23:
Tried several times with start Button After waiting some seconds always the
wrong values appear.
I think it is the time, till all data are scanned.
The results when wrong are always the same way wrong and for example if
cell temperature rises Data change also frequently. So there must bei a
connection when the First 3 cell temperatures seem always ok and the
following are always the same way wrong and change the same way wrong . Is
there an additional data when activated ore any shift of Data?
How can received Data be logged for your evaluation.
Because of no Informations for months is the project stopped?
Always waiting till ZOE sleeps is not a useful solution.

yoh-there, 2018-03-23:
The project is not stopped. But it is 99% certain a hardware problem with the latest KONNWEI dongles (as in: can't handle the extra bus load). Reason of slow progress is:
- no time
- no 90 model car available to test
- dongle market is a mess, almost everything is "no brand" so impossible to give decent buying advice.

R240Pe, 2018-03-23:
Thanks for quick answer. There is no criticism, you have done a very good job!
I would even buy 10 dongles if this is certain to solve the problem properly  ... to read correct Data. 
As described above on Dec 15 2017: Bus Speed on Q90 is expexted to be the same as R240. My Dongle works fine on R240. You can say it's fast enough.
I controlled many screeshots from Q90 CanZE - it can be determined, always the same Values are correct, whereas the others are always wrong or missing. 
Thus I am shure random errors can be excluded. 
It is always the same returning cause after pressing the Start button or starting charging.
How can I get a logfile from CanZE of all collected/delivered data from Konnwei 
- first: when ZOE is awakend without pressed Start button
- second: when additionally Start button was pressed.
A comparison could show the differences if and where there are any.

yoh-there, 2018-03-23:
Thank you, not perceived as critisism at all.
Bus speed (bit speed) is the same, but it seems the bus LOAD is different.
Logging is done entirely after the dongle of course. Can be configured in the setting screen. I think there are a few open issues for logging (not involved myself)

WLBI, 2018-08-06:
Is there already an update? I have the same problem with the Zoe Q90.

fesch, 2018-08-06:
No ... not yet ...

yoh-there, 2019-01-14:
@StephaniHi commented:

> Just did my first charging session with my R110 and noted all the issues already described here. I watched crazy values in the heatmaps for more than an hour without them turning to something useful.
> Max Battery recharge is shown as 0.0. Real SOC is shown as 0.0.

Can I ask if the problems were over when the car was off the charger and NOT started?

yoh-there, 2019-01-28:
Closing due to lack of follow up. Mose will be rolled up by #504 and the CanSee dongle

closed 2019-01-28 by yoh-there

matthew1471, 2019-02-03:
I had crazy values before.. then I found the dongle's guiding plastic was larger than the yellow Zoe guide... dongle used to get stuck to the yellow plastic a lot and come up.. eventually yellow plastic broke and came out with dongle.. I unattached the yellow plastic and inserted the dongle again and it made contact with the pins better..

Weird.. NaN temps and other CanZE errors all but disappeared in my R90.

So if you're consistently getting weird errors in certain CanZE functions (temp heat maps was my main one), check you've got good connectivity to the car..

It was a weird flaw as it was connecting and pulling some data.. just weird data to certain requests.. either certain address ranges go over certain pins or it was particularly large packets etc.

CanZE works fine on R90 for me now and I doubt Q90 would be significantly different.

yoh-there, 2019-02-17:
Late reply but thanks for that Matthew. Indeed voltage heatmaps are large messages coming in fast (though it's still all packetized in 8 byte frames)