fesch/CanZE #474: iOS: CanZE stopps communicating

leckermojito, 2017-11-06:
After about 1 Minutes the CanZE App stopps updating the screen and does not get any data from the OBD Dongle. The LEDs on the Dongle are dark after the Freeze. With other ODB Apps the dongle works flawless. 

A restart of the App resolves the Problem for another 1 Minute and the App stopps again. 
iOS 11.1 on iPhone 7Plus

danhart, 2017-11-07:
I’m also experiencing this same issue. Sometimes it does stay connected for longer, but mostly it stops communicating after about 1 minute. Happy to provide any extra details you require.

FredLeudon, 2017-11-12:
What kind of dongle are You using ?

Wifi ? Bluetooth ?

Frédéric RICHARD

leckermojito, 2017-11-12:
Im using a Konwei Wifi Dongle. As far as i know a bt dongle can‘t be used with iOS. The dongle works fine with other types of OBD2 Apps also over a long period of time. Thats the reason why i am asume that the dongle is not the problem

FredLeudon, 2017-11-12:
Does CanZE app stay on foreground or do you use it in background ?

I will have a look at this issue. 

Frédéric RICHARD

leckermojito, 2017-11-12:
I have never tried it in background. Always in foreground mode. 

Thanks for your work

FredLeudon, 2017-11-12:
Strange, I’m using iPhone 7 too and iOS 11.2 beta and i don’t have this kind of problem.

Ça you clear logs on CanZE app, kill app (double clic on home button and swipe CanZE app up) laugh it again, use it until it stops working, kill it again, and click to « send feedback »

Thanks

Frédéric RICHARD

leckermojito, 2017-11-14:
Hi Frederic, 

strange behavior: the problem disappears. On my last test i was 100% sure that i had the 3.0 Version and it stopped after 1minute or so. Test today morning was without any issues.

cheers

Ben

closed 2017-11-14 by leckermojito

FredLeudon, 2017-11-14:
Cool ;-)

Frédéric RICHARD