fesch/CanZE #476: iOS: no data communication with the Twizy under different iOS versions possible

betaa317, 2017-11-11:
Hello, we are a student team from the TU Munich and are trying to improve the thermal management of the Renault Twizy. While researching we came across CanZE and we believe it could help us a lot to understand the Twizy better.

We tried to use the app with an iPhone 6 with iOS 11 and the iCar 2 WLAN OBD Dongle and could establish a connection with CanZE (and also EOBD Facile), but no data was shown and an error message appeared saying the data does not exist.

We also tried it with an iPhone 4S with iOS 9.3.5 - a connection was also possible and we could see the values of the cell voltage and velocity and some other values that did not make sense but nothing else (see attached pictures). The error message still appeared.

We found some forum entries online of people who were able to use CanZE with the same dongle we used, so that should not be the problem. 

Do you have any idea what we could do to make CanZE work properly with our Twizy and make a data communication possible? This would help us a lot!

Thanks in advance, we are really hoping you can help us with our issue.

Elena

[image]
[image]
[image]
[image]

yoh-there, 2017-11-11:
At the moment we have no code for the Twizy, but I know @fesch  has been investigating with a local Twizy driver and maybe there is some progress he can report. Twizy has some subsystems that are similar, but also some that are vastly different. I bet the BMS is one of the more different ones.

I know though there is massive knowledge, including even "chip tuning" the Twizy in "the internet". The OMVS project did quite a bit of work and I have a contact there. I will ask around a bit and/or look for code.

If you are a bit into programming it is not too hard to get basic communication going, the issue is more the semantics of the messages. Please consider getting an IT student in your team if not already.

yoh-there, 2017-11-11:
PS: please share an email address I can use. I do have some useful information, but I don't know if I can post it in public.

FredLeudon, 2017-11-11:
Hi,

I’m working on a CanZEv2 app.

I need beta testers. 

If you are working on twizy, I’ll be glad to add You to the beta-testers team.

Please, send email address to ioscanze[at]icloud[dot]com

Yet I’m very busy at work and at home (twins are born in June) and project is moving slow.

But I hope being able to sent a first beta version in January.

Regards

Frédéric RICHARD

fesch, 2017-11-11:
The Android Version works for Twizy, but as the PID's differ from the one used in the Zoé, not all data is visible or displayed.

betaa317, 2017-11-14:
Thanks for your answers! If you have any further useful information for us, you can also contact us via our email address globaldrivetwizy@gmail.com :)

FredLeudon, 2017-12-18:
CanZE v2 is still under development. 

If your agree, I can add your email address to beta testers.

I have no Twist to test so help will be useful.

At this time, only Battery-Temp is ok.

betaa317, 2017-12-18:
Hi Mr Richard,

thanks for your update. Of course you can add our email address to the beta testers - maybe you can add our official email address globaldrivetwizy@gmail.com <mailto:globaldrivetwizy@gmail.com>, so everyone in our team has access to the information.

We hope to get a Twizy later this week or in the beginning of next year and we will be glad to help as soon as we can! And we are happy with every information you can give us, even if it is just the battery temperature, this might help us already.

Thanks in advance, we are looking forward to hearing from you again.

Kind regards

Elena Weisbrodt

yoh-there, 2019-01-28:
Closing due to lack of activity

closed 2019-01-28 by yoh-there