fesch/CanZE #478: charging history always shows slow

nean-and-i, 2017-11-28:
Hi,

I have a used Q210 / 2013, and when I readout charging history with CanZE it always shows "slow" never quick or fast !?

It's very stange because I charged the car at the airport nearby with super charge stations that are able to charge cars up to 50kW (50 kW DC 400V 125A, 43 kW AC 400V 63A)

Well, I took the readouts serious, thought there is an issue with the car/controller and went to Renault to get the car checked and confirmed that the car is able to do quick-charge.

Renault confirmed they checked everything and all is OK.
I was not able to get a definitve answer of the term "quick charge" 
The car was carging on 3-phases, at around 38kW constantly which they said this is very good.
But the readout of the charging history classified it as "slow"

Which parameters are used in CanZE to clasify charge types?
Is that in line with Renault's understanding of quick-, fast- and standard charging cassification?

Can someone please demystify quick-charge ?

According to Renault
quick charge: 3 phases, 400V, 43kW, 80% in 30min
fast/accelerate charge: 3 phases, 400V, 22kW, 80% in 1h
everything below is a "standard charge"

thanks,
nean

yoh-there, 2017-11-28:
It's what in the technical documentation. That is all I can say about it. Drawing 38 kW you certainly were quick-charging! Better look at charging screen and charging graphs screen to get an idea how the charger is performing.

And here is some data of what you can expect given battery temperature and State of Charge.

https://docs.google.com/spreadsheets/d/1zDhdxQWK0W7SLOC40KOZc_lyTW6vkk5KnUB6xARV55I/edit?usp=sharing

closed 2017-11-28 by yoh-there

nean-and-i, 2017-11-28:
That would then mean the charging history readout of CanZE is not correct?
If 38kW is quick-charge then I'd expect to have it in the history rather than stating "slow" ?

yoh-there, 2017-11-28:
I would prefer the notion that the car does not provide us with the proper info / definitions :-)

nean-and-i, 2017-11-28:
I asked the Renault engineers if they can read out the charging history to proof the car is able to do quick charge, but they said this is not possible, not even with original Renault equipment.

Are you guys sure that what you suppose to read out is the "charge type" ?
Is anybody out there who has other values than "slow"?

Is there a bug in CanZE or in my car?
If the car does not provide proper or invalid values, can you guys please use a different default value so people does not get confused?