fesch/CanZE #484: Mean Battery Temperature has no decimals

wolfijenne, 2017-12-18:
The mean Battery Temperature displayed in the charging screen (and elsewhere I guess), is missing the decimal.
With 
7bb,56,71,10,0,0,°C,2103,6103,5\n" // Mean battery compartment temp
I was able to get a decimal value. I would assume that the float conversion is missing (as I had the same problem too).
i.e.
Display CanZE: 4°C
Value: 4.8°C

yoh-there, 2017-12-19:
For all who read this: this is only for Fluence and Kangoo. Took me a while to realize. ZOE does not have that field in the LBC.

Next, I must confess I am surprised it even works, as the resolution seems to be set at 10C instead of 0.1C. It would be helpful if you added which changes you exactly made please.

wolfijenne, 2017-12-19:
Oh actually I use this definition on the ZOE (please note, that I edited the issue before). I am working on the OVMS and stumbled over this there. The relevant code over there is 
`float(CAN_7NIBL(44))-40)/10`
while CAN_7NIBL is a macro to extract the 7 Bits out of the middle

yoh-there, 2017-12-19:
Ah OK, that explains at least the resolution 10 issue!

If you are using this on ZOE could you please go to the firmware screen and report the 4 firmware lines of both the LBC and LBC2? I am asking because I have a completely different field definition for 7bb.6103 (see actors\EcuDiagLBC.java); I might have loaded improper code there, or we're bouncing into two rather different LBC version. Thanks.

wolfijenne, 2017-12-19:
I'll check tomorrow... However, please note that I use 
42e,44,50,1,40,0,°C,,,e3\n" // HV Battery Temp
not 7bb... 
(I mistakenly reported first the 7bb line, but corrected it later)

yoh-there, 2017-12-19:
OK, thanks for the clarification, I was scratching my head ;-)

The number in the 42e frame is a 7 bit number with a resolution of 1 degree. This came from formal documentation. While I can't see an error in your code snippet, you should not divide by 10. It would be very strange if that was correct, because the range would then be only 12.7 degrees. It actually runs from -40C (value 0) to 80C (value 120).

wolfijenne, 2017-12-19:
Thanks for that input. I'll double check my values with other real Battery temperatures then. Your argument kind of rocks ;)

yoh-there, 2017-12-19:
Great :-P ! We're really always open to input. I will close this issue now, but you can post your results, it's always appreciated.

closed 2017-12-19 by yoh-there