fesch/CanZE #486: Power figures sometimes missing negative sign?

jinfenggoh, 2018-01-08:
Hi CanZE team, 

Thanks for the great work so far. I have been using the app a lot to try and get some representative consumption numbers on my R240 to try and see if this data can be used to inform driving speed. 

To do this, I put the car on cruise control and hope for a good flat stretch of road. 

The data that is recorded seems to show quite a lot of variation (see graph). However, I believe that many of the positive motor power figures are actually data being recorded without the "-" sign. This is because the figures seem to be more or less the same, which I would expect, but just in the positive side of the graph rather than the negative side. 

See attached graph from Datplot: 
[image]

Do you think this could be the case? If yes, is it a known bug? 

Thank you for your time and I wish you all the best in the new year, keep up the great work and know that we, the drivers, are all incredibly grateful for the team taking the time to work on this. 

Best regards,
Jin Feng Goh

yoh-there, 2018-01-08:
This doesn't look right at all. Power should be relatively stable, for 100km/h on a relatively flat road in the order of 15-20 kW. How was this data gathered?

jinfenggoh, 2018-01-09:
Hi Yoh-There, 

The data was logged in the Sony Xperia M5 using a OBDII 902 scanner. Scanner looks like this: 

[image]

My CanZE setting page looks like this: 

[image]

and the resulting log file: 

[data-2018-01-07-17-44-00.log](https://github.com/fesch/CanZE/files/1614687/data-2018-01-07-17-44-00.log)

One of the clearer figures is in "2018-01-07-18-39-10" until "2018-01-07-18-39-20" with data at "2018-01-07-18-39-15" being positive while the other 2 are negative. 

Also interesting is that the power figure is, I assume battery V × battery A, but for the data on 18:39:15, the battery A is negative, but the resulting power figure is positive.

I am also wondering if you might be happy to enlighten me on the meaning of the column just before the "Range" column. I have not been able to figure this out yet... 

Thanks!

Kind regards,
Jin

yoh-there, 2018-01-09:
Thank you Jin for the details.

@fesch maybe you need to have a stab at this as it is the logging function?

yoh-there, 2019-01-13:
Closing due to lack of activity. Feel free to reopen.

closed 2019-01-13 by yoh-there