fesch/CanZE #488: Android Auto

DiLeGreen, 2018-01-21:
I would like to request Android Auto or Mirrorlink Support, so we can have CanZE on the R-Link Display.
Maybe this can be archived by the use of the Developer Mode of Android Auto, like OBD2AA ( https://github.com/borconi/obd2aa ) does for Torque on the Ioniq.
Maybe http://www.thekirankumar.com/blog/2017/12/29/play-youtube-video-android-auto-app/ or https://github.com/thekirankumar/youtube-android-auto/ can also be a help.

yoh-there, 2018-01-22:
@fesch What do you think about the Andoid Auto rave, now that the worms are out of the can with the latest "hack"? I did a quick read up and while I fancy screen casting myself, it doesn't seem AA is "intended that way". Maybe I am misreading.

fesch, 2018-01-22:
Screen mirror the phone's screen onto the car's display would be really great and solve my problem of having another device fixed somewhere in the car, but I'm am not sure if we really should encourage people to do this hack....

landswellsong, 2018-06-13:
@fesch @yoh-there my 5¢. The way I see it, they made a whitelist of the apps that are able to screencast, but it's definitely possible to be on that list, as seen by having Waze. I think there is an application procedure for this, but Google requires a few guidelines about the UI:
https://developer.android.com/docs/quality-guidelines/auto-app-quality
https://developer.android.com/distribute/best-practices/launch/distribute-auto

The hack described above can work for testing (it's still working given the xda-developpers thread on the subject), but there may be a way to make it available for general public if Google admits the app in the whitelist.

I'd like to volounteer for the AAuto SDK part, but I have like literally zero experience in Java (C/C++ coder here) and I'd need a hand with the "make the UI comply with what Google wants" part.

You may remember me from the comments about Zoe with Nissan Leaf battery (actually it's installed and working now) and getting CanZE to work on the car's screen would be a great help here since the BMS firmware is still in development.

fesch, 2019-01-28:
I agree on that ...

closed 2019-01-28 by yoh-there

landswellsong, 2019-06-19:
>Unless somebody really is inclined to pick up the challenge

*raises hand*

I've played with the android auto support jar, sounds like I could do this on a hobby basis slowly, but from what it seems to be now (disclaimer: not an Android coder per se), the activities displayed on the car need to have separate activity classes deriving from a different hierarchy than the current phone ones. I.e. for code reuse most of the stuff needs to be converted to fragments and then pulled into 2 separate activities. Is that a feasible thing that can be done?

Other thing is that the jar is supposedly coming from a dubious source, I guess there is some diplomacy to be made with Google for them to provide the official one (after meeting the criteria)

landswellsong, 2020-11-20:
@fesch @yoh-there hey-hey

https://android-developers.googleblog.com/2020/10/introducing-android-for-cars-app-library.html

yoh-there, 2020-11-20:
Fun, thank you!