fesch/CanZE #489: Kwh consumed

FrancescZOE, 2018-01-22:
Hello from spain again :-) 

CanZE version which can read Kwh consumed in my Q210 from 2013 and old mobile with old version is version 1.24  Build 2017.02.21 @ 14:40

[image]
[image]
shown as energy consumed :-)

FrancescZOE, 2018-01-22:
It would be really nice to see this feature at least in experimental. That way i can control exactly kwh with 0.1kwh acuracy in any travel. ZOE consumption r-link is so bad :-(
There is also another feature i can read in iCarSoft odb2 reader. Kwh total in recharging jejeje

yoh-there, 2018-01-22:
Thank you for providing this info. I double checked 1.24 and the definition is the same as in my (defunct) in experimental. I will do some further experiments later, as right now it does not work.

FrancescZOE, 2018-03-01:
Hello please put the code in new versions at least in experimental side. In my new mobile i cant see kwh consumed but i do in my old mobile with that 1.24 version installed...
There is also another KWh data about " energy during recharge"  IT seems it is really the lecture of kwh used by AC to charge at DC. In another odb reader i can get at that point. How could i help you to implement that? perhaps a photo ot if?

yoh-there, 2019-01-28:
Closing this issue. Total energy taken from the grid is a totally different number, and I will have a separate look at that in the future to see if it's available, see #519 . Trip energy used (so after charger and battery inefficiencies) is going to be implemented as described in #513

closed 2019-01-28 by yoh-there