fesch/CanZE #490: "Energy consumed kWh" on the Charging screen - issue with Q90

kiwizoe, 2018-02-05:
Our family has two Zoes, a Q90 (2017) and a Q210 (2015).

I've noticed that under "Energy consumed kWh" on the charging screen the value is always 11185 kWh, and never changes. Whereas in the Q210 it does change and reads a sensible value.

Ie, it all seems fine for the Q210 but there seems to be a bug for the Q90.

Please see the attached screenshots.

Version number: 3.0.0.6, iOS
Car: Zoe Q90 (2017) & Zoe Q210 (2015)
Dongle: OBD II Wifi adapter

[image]
[image]

yoh-there, 2018-02-05:
i think this should be combined with #489 We've removed this field in the Android version a while ago as it was problematic. In my q210 it does not show at all.

bjaolsen, 2018-11-09:
I use Android and it still shows up, however since I am on 89.000 km it now only shows 9999 kWh since I am over.
ddt4all shows somewhere around 15.000 kWh for the battery

yoh-there, 2018-11-12:
The field we use is a 24 bit field and is given in Wh. THis means a maximum range of 16777 kWh. I cannot match this with "9999".

The field id is 2233dc on the EVC, bits 24-47. I assume it is a different field than ddt4all. If you can see the id that is used, I'd appreciate it.

yoh-there, 2019-01-28:
Closing due to lack of activity. Rolling up into new issue #519

closed 2019-01-28 by yoh-there

matthew1471, 2019-02-02:
> 
> 
> The field we use is a 24 bit field and is given in Wh. THis means a maximum range of 16777 kWh. I cannot match this with "9999".
> 
> The field id is 2233dc on the EVC, bits 24-47. I assume it is a different field than ddt4all. If you can see the id that is used, I'd appreciate it.

Renault DDT2000 Database is in here : https://docs.google.com/document/d/1IBDlq_Ten-jkCF9zVC-n9ISZchUngfiJ9JrrYgPVG1I/edit and the Zoe is "X10"?

The proper values for the fields for instance are listed in X10_X98_X87_X52ClimBoxGen1_v08_3_20160722T152536.xml

matthew1471, 2019-02-03:
Ah my bad, X10 appears inside more than just that XML file :)