fesch/CanZE #491: I cannot get out of "Safe Driving Mode" (iOS)

kiwizoe, 2018-02-05:
I cannot get out of "Safe Driving Mode" (iOS)

How to replicate:
1. Activate safe driving mode (see screenshot 1)
2. Deactivate safe driving mode
3. A pop up message appears (see screenshot 2)

Tapping on 'Yes' or 'No, thanks' does not close the pop-up box.

The only way I have found the way to move on is to close the CanZE app, which is all a bit drastic. And I'm still in Safe Driving Mode.

Version number: 3.0.0.6, iOS
Car: Zoe Q90 (2017) 
Dongle: OBD II Wifi adapter

[image]
[image]

leckermojito, 2018-03-11:
Issue #475 is the same topic.
I see the same problems.

moms-spaghetti, 2018-04-19:
Hi is there any progress on this? I too am experiencing an issue when trying to disable safe driving mode on iOS. The application halts and has to be closed and restarted. Thanks

moms-spaghetti, 2018-04-20:
Just checked my last suggestion and sadly even with location services enabled the app still crashes when trying to disable safe driving mode.

kiwizoe, 2018-04-20:
@moms-spaghetti thanks for the thought though, good thinking!

moms-spaghetti, 2018-04-21:
Thank you Frédéric. Do you know if there is any workaround for disabling ‘safe driving mode’ in the mean time?

I hope the Apple review doesn’t take too long.

FredLeudon, 2018-04-21:
Sorry, no idea. 

Frédéric RICHARD

kiwizoe, 2018-04-21:
> I will try to send an update to Apple store next week.
>
> Don’t know if it’s possible because I already send v4 beta.
>
> I can’t release v4 yet because I have to redo lots of screens (v4 is redeveloped from scratch).
>
> Frédéric RICHARD

Hi Frédéric

Thanks so much for all your efforts. 

If you’d like me to help beta test via TestFlight, now or in the future, only too happy to help. Just let me know. 

Best wishes
Mark

FredLeudon, 2018-04-22:
Yes, send your Apple ID by mail at : ioscanze[at]icloud[dot]com and I will add you as beta tester. 

Frédéric RICHARD

yoh-there, 2019-01-28:
Closing due to lack of activity

closed 2019-01-28 by yoh-there