fesch/CanZE #492: "Max AC charge pilot" seems to be incorrect when charging on three phase

kiwizoe, 2018-02-05:
When charging the Zoe on single phase, the "Max AC charge pilot" matches the current that I have selected on my EVSE.  I assume that this is reporting the CP (control pilot) setting.

However, when I charge the Zoe on three phase, the "Max AC charge pilot" is one amp less compared to that which I have selected on my EVSE.

The problem could be with my EVSE. But is it possible that there is a bug that only shows when charging on three phase which results in the value presented here being 1 amp smaller than it should?

Version number: 3.0.0.6, iOS
Car: Zoe Q90 (2017) 
Dongle: OBD II Wifi adapter

[image]
[image]

fesch, 2018-02-05:
This is "normal". What the ESVE offers is the maximum the car can take, but usually on 3 phases it doesn't got to the limit.

closed 2018-02-05 by fesch

kiwizoe, 2018-02-06:
Are you sure? Isn't this number the Control Pilot (CP) signal coming from the EVSE? I assumed this as (1) it's constant during the charging cycle (2) the label "Max AC charge pilot" I thought may have been a translation error (ie charge pilot for control pilot).

But perhaps this number is something else?

yoh-there, 2018-02-06:
This interpretation of the meaning of the CP signal is coming directly from the ZOE. I have often noticed a 1 amp difference, also on 1 phase. Not much we can do; it's what the car reports. It's not like we're measuring the pulse width of the CP.

kiwizoe, 2018-02-06:
Thanks for the explanation - that makes sense.

By the way, thanks for all your efforts with CanZE, I've really enjoyed using it.