fesch/CanZE #493: Source Code iOS

beseder42, 2018-02-06:
Is the source code for iOS also available als open source? Or is that intended in the near future?

FredLeudon, 2018-02-06:
CanZE for iOS is made with PCSoft’s WinDev Mobile app. 

I can share source code but it’s not easy as github’s repository.

Frédéric RICHARD

beseder42, 2018-02-06:
If you can share I would appreciate.

yoh-there, 2019-01-05:
Does Windev still uses it's own Source Code Manager?

But even without git integration, if the source is text files (like C-code, XML resources, etc etc) I would highly recommend creating a git repository.

Anyway, if there is not a shared location of the source maintained or planned, I respectfully propose to close this issue.

yoh-there, 2019-01-13:
With some sadness, closed due to lack of activity.

closed 2019-01-13 by yoh-there

FredLeudon, 2019-01-14:
Windev uses it’s own editor. 

Source code is not text file, so creating git repository is not easy.

Frédéric RICHARD