fesch/CanZE #494: String sorting in the battery health table

landswellsong, 2018-02-27:
Hi,
I'm not sure I have to report it here or elsewhere, but I noticed that the tables in battery health survey use string sorting. This makes sorting by the battery health unusable. I realize those fields also have text to them, but maybe there is a way to sort them numerically based on the first word / number sequence up to non-number etc.

yoh-there, 2019-01-13:
Sorry for not responding. This seems to be an issue with the table handler in Wordpress (or we should enter all data with leading zeros).

I am closing this issue but not without suggesting to @fesch to ditch the surveys? Is anyone doing something with the data?

closed 2019-01-13 by yoh-there

fesch, 2019-01-13:
I will take a look at it, but the table contains strings, and this cannot be changed right out-of-the-box.