fesch/CanZE #500: Battery Available Energy not working on Kangoo ZE (2013)

jvalik, 2018-06-06:
Hello,

first of all I would express many thanks that you developed CanZE tool. It helped me to check the battery health and cell voltages at the time I was purchasing my Kangoo.

My Kangoo ZE was produced 5/2013 and I'm having trouble reading Available energy of the battery and few more related parameters. It looks like some datastream isn't available or so.

I tried with ingition both on and off, having the car charging plugged and not, but it still doesn't work.

Sometimes I receive timeout on [atm] messages, sometimes not. I was receiving it also on cell voltages chart, but turning off the Safe Driving Mode solved that issue - not sure why.

Attached are the settings screenshot of CanZE and the extended Charging screen. I'm using Konnwei KW902 bluetooth dongle.

I'm aware that developers have Zoes only, so I can offer some help with Kangoo diagnosis, for example going through some raw data from OBD etc. But I'm new to OBD and not familiar with it, so some basic instructions may be needed.

Thank you

John
[image]
[image]

jvalik, 2018-06-07:
Hello,

I also tried to run Fluence ZE Spy on my Kangoo and it was able to show much more data, see attached screenshot. Hope it helps.

John
[image]

fesch, 2018-06-07:
I can conclude, that the Fluence PID's are closer to the Kangoo as the Zoe's are ;)

yoh-there, 2018-06-07:
I am with @fesch on this one. AFAIK a Fluence ZE and a Kangoo ZE are technically the same.

yoh-there, 2019-01-13:
@avataranedotas if you feel up to it, would you care to have a look at this? For the fields showing no or crazy data in the OP's second screen print, if you have the usual definition I (he) would be grateful. I can shove it in the definitions if I have it.

avataranedotas, 2019-01-18:
I currently don't have time to look into it. I'm busy with the "new" 2017 LG battery PID's for the Fluence ZE.

yoh-there, 2019-01-18:
Fair enough! Given lack of activity, I am closing this issue.

closed 2019-01-18 by yoh-there