fesch/CanZE #504: Is it possible to translate free frames to iso-tp frames

FredLeudon, 2018-09-21:
Hi,

As you perhaps know, I am building an brand new CanZE for iOS (it will be 4.0.x)

Actually all features are not redeveloped.

I saw the excellent Ddt4all program and I decided to scan vehicle ECU to get the right definition.

Actually it is under beta test and it works fine with Zoe. Twizy and Kangoo virtually works too but I’m waiting for beta testers feedback.

I realized too that querying free frames causes a lot of timeouts.

Please, CanZE team, help me convert free frames to iso-tp frames ;-)

I have actually trouble with torque (wheel, motor and break)

Thanks a lot

yoh-there, 2018-09-21:
I will dig through my documentation tonight or tomorrow.

Going ISO-TP is of course much more reliable as it is a question-and-answer game. However, it is actively engaging the car, whereas free frames are just sniffed upon.

So my personal view is that free frames are actually preferred from a technical point of view, but if our goals is still to do it with crappy dongles, ISO-TP is the only way.

Compromises, compromises.............. ;-)

yoh-there, 2018-09-21:
Fred, as an experiment, can you try to replace

+ "130,44,55,-3,4094,0,Nm,,,ff\n" // DriverBrakeWheelTq_Req

With

 +"7bc,28,39,1,4094,0,N·m,224B7E,624B7E,ff,Driver brake wheels torque request\n" //

And change the corresponding  code to use

7bc.624b7e.28 (instead of 130.44)

If this works I can try to find more equivalents.

FredLeudon, 2018-09-25:
thanks for your answer.

The fields I can use in my canZE are below : 

//[77E.223027.4.0] : ElecMachineMaxGenTorque - Unit[Nm] [*** NO DATA ***]
//[77E.223026.4.0] : ElecMachineMaxMotorTorque - Unit[Nm] [*** NO DATA ***]
//[77E.223024.4.0] : ElecTorqueRequest - Unit[Nm] [*** NO DATA ***]
//[77E.223025.4.0] : PEBTorque - Unit[Nm] [*** NO DATA ***]
//[7BB.2103.20.0] : 21_03_#20_MG_Torque - Unit[Nm] [*** NO DATA ***]
//[7EC.222245.4.0] : Limited electrical motor (EM) effective torque setpoint - Unit[N.m] [*** NO DATA ***]
//[7EC.2233F3.4.0] : Maximal torque without unballasting - Unit[N.m] [*** NO DATA ***]
//[7EC.2233F4.4.0] : Minimal torque - Unit[N.m] [*** NO DATA ***]
//[7EC.222247.4.0] : Minimum effective torque that can be requested to the electrical motor (EM) - Unit[N.m] [*** NO DATA ***]
//[7EC.222241.4.0] : MSR torque setting sent to torque arbitration - Unit[N.m] [*** NO DATA ***]
//[7EC.223463.4.7] : electrical braking torque setpoint - Unit[] [*** NO DATA ***]
//[7EC.222246.4.0] : Electrical motor (EM) maximum effective torque available - Unit[N.m] [*** NO DATA ***]
//[7EC.223448.4.0] : Electrical motor (EM) maximum effective torque available, after eco mode limitation - Unit[N·m] [*** NO DATA ***]
//[7EC.22223B.4.0] : Electrical motor (EM) torque setpoint before AJS Correction action - Unit[N.m] [*** NO DATA ***]
//[7EC.223044.4.0] : Estimated torque of the electrical engine (from PEB) - Unit[N.m] [*** NO DATA ***]
//[7EC.222243.4.0] : Final effective torque request to the electric motor (EM) - Unit[N.m] [*** NO DATA ***]
//[7EC.22223E.4.0] : Final torque setpoint after AJMOD action - Unit[N.m] [*** NO DATA ***]
//[7EC.222249.4.0] : Force demand (at the wheels) when powertrain setpoint is equal to zero, before saturation accounting for minimum torque demand available - Unit[N] [*** NO DATA ***]
//[7EC.22223D.4.0] : Torque correction calculated by AJCOR function - Unit[N.m] [*** NO DATA ***]
//[7EC.223043.4.0] : Torque request - Unit[N.m] [*** NO DATA ***]
//[7EC.223465.4.0] : UBP wheel torque request from CAN - Unit[N·m] [*** NO DATA ***]
//[7EC.22223C.4.0] : ASR setpoint torque sent to torque arbitration - Unit[N.m] [*** NO DATA ***]
//[762.22011F.4.0] : DID - Assistance torque demand - Unit[Nm] [*** NO DATA ***]
//[762.220112.4.0] : DID - Driver Torque - Unit[Nm] [*** NO DATA ***]

I just made an specific activity to record all of them.

 I will test it this afternoon while going home.

yoh-there, 2019-01-13:
Bumping this issue. Any progress on this?

Personally, I am on the fence on this one. 

1. For shitty dongles, trying to fetch everything through diagnostics commands makes sense, but it's slow and would require a huge investment in time to go over all fields we use, both to find all the equivalents as well as going over all code and layouts.
2. For the new CanSee dongle it's not that important, but what can be fetched as free frame does not have any influence on the car and is a lot faster too.

yoh-there, 2019-01-13:
As a result of this discussion I have bitten the bullet and recoded all the data definitions from strings in the Java code to csv files in the assets. Nothing changed to the internal structures, just how they are initialized.

This way it is probably much easier to share those definitions between iOS and Android. @FredLeudon you might want to consider to use the same format, and if better, I am open other formats. Also, we might consider an external "overrule" file to do quick experimentation.

As I am now pretty busy with the CanSee dongle I won't be doing any redefinition as described above soon. I would prefer to
- agree together on a shared file format for Fields, Frames, Ecus, DTCs and Tests
- possibly add our own "CanZE functional ID", so the code can refer to that instead of hard frame ids
- if keeping long descriptive "static final SID variables", move them to one place
before moving on.

Just for reference: we only use 44 free frame definitions in CanZE

yoh-there, 2019-01-14:
Copying from issue #510 
==========
• establish data connection to the dongle (BT or WiFi and CANSee when my diy dongle will be operational)
querying ECUs to read vehicle’s VIN
once VIN is read, I can load ECU address array* 
each address is scanned to retrieve ECU version numbers*
ecu definition is downloaded from my server*
these definition is used to load Request, data items and data arrays.
operations are done only once and ecus version are stored locally in database files. 
==========
@FredLeudon  It seems we're heading down different paths. Going back to the original issue: I assume you are using the DDT4All XML database. Have you made a translation of the 44 free frame fields and if so, can you share them?

FredLeudon, 2019-01-15:
Hi, sorry for late answer, i’m verry bussy at work ...

I dit not translate free frames. 

You’re right, i use ddt2000 xml files. 

First, I build a database file with each ecu version number (soft, supplier, address, etc)

I used to analyze the ecu definition located on ecu directory but it was not correct. Some ecu are not referenced. So I decided to bold this database file by analyzing all xml files.

To have a good database, I downloaded different releases of ddt2000 database (hard to find recent) I merged.

I uploaded xml files to my CanZE server. 

The approach is to read VIN, determining project (x10 for Zoe), loading list of addresses used by the project (I have my own database file too) and querying each address with identification request to get supplier, version, soft, ...

I wrote (inspired from ddt4all) a check ecu function that try to find the exact or more closer ecu definition.

Then I download xml file from server, analyses it and create local database files : ecu, data, requests, devices, dataitems.

Then I used these database files to build a ecu class, added to a array of ecu class in a vehicle class. 

The database type I use is Hyper File, WinDev’s database.

If you want, I can export windev files to xml or json files on my server, and give access to the CanZE team.

I will give you the Sid I use in CanZE for iOS asap (now, some Sid for some projects are wrong)

yoh-there, 2019-01-15:
Hi Fred,

Yes I would be interested in the the translations you have made so far for the 44 fields. Errors is no problem, better I put some energy in correcting those than starting over. I did a few earlier today, but as you realize, it's tedious en boring work.

What I will do is make an "alternative" set we can compile to test if we can then support el cheapo lousy diagnostics-only dongles.

FredLeudon, 2019-01-17:
Hi,

I did not yet translate ALL free frames to ISO-TP frames.

Below there are the frames I use in CanZE for iOS

:tabSID["SID_VERSION"]								=	[["NUMERO","12"],["DATE","20181231"]]

:tabSID[SID_Vin]									=	[["*ALL","763.2181.3.0"]]

//														X07 : Twingo						X09 : Twizy							X10 : Zoé							X38 : Fluence						X61 : Kangoo 2 ph1					X61Ph2 : Kangoo 2 ph2						
:tabSID[SID_Odometer]								=	[["x07","7E8.222006.4.0"]		,	["x09","763.2118.3.0"]			,	["x10","7EC.222006.4.0"]		,	["x38","7E8.222006.4.0"]		,	["x61","763.2113.0.0"]			,	["x61Ph2","7E8.222006.4.0"]			]
:tabSID[SID_HVBat_SOH]								=	[["x07","7EC.223206.4.0"]		,	["x09","7BB.2161.11.0"]			,	["x10","7BB.2161.10.0"]			,	["x38","7EC.223206.4.0"]		,	["x61","7BB.2161.11.0"]			,	["x61Ph2","7EC.223206.4.0"]			]
:tabSID[SID_HVBat_SOH_Min]							=	[["x07","7BB.2161.11.0"]		,	["x09","7BB.2161.11.0"]			,	["x10","7BB.2161.11.0"]			,	["x38","7BB.2161.11.0"]			,	["x61",""]						,	["x61Ph2","7BB.2161.11.0"]			]
:tabSID[SID_HVBat_Temp]								=	[["x07","7BB.2104.*;9,2,67"]	,	["x09","7BB.2104.*;5,3,23"]		,	["x10","7BB.2104.*;5,3,74"]		,	["x38","7BB.2104.*;5,3,74"]		,	["x61","7BB.2104.*;5,3,14"]		,	["x61ph2","7BB.2104.*;5,3,74"]		]
:tabSID[SID_HVBAT_Temp_Min]							=	[["x07","7BB.2104.5.0"]			,	["x09","7BB.2104.75.0"]			,	["x10","7BB.2104.75.0"]			,	["x38","7BB.2104.75.0"]			,	["x61","7BB.2104.15.0"]			,	["x61Ph2","7BB.2104.75.0"]			]
:tabSID[SID_HVBAT_Temp_Max]							=	[["x07","7BB.2104.3.0"]			,	["x09",""]						,	["x10","7BB.2104.77.0"]			,	["x38","7BB.2104.77.0"]			,	["x61",""]						,	["x61Ph2","7BB.2104.77.0"]			]
:tabSID[SID_HVBAT_Temp_Avg]							=	[["x07","7BB.2104.7.0"]			,	["x09","7BB.2104.76.0"]			,	["x10","7BB.2104.76.0"]			,	["x38","7BB.2104.76.0"]			,	["x61",""]						,	["x61Ph2","7BB.2104.76.0"]			]
:tabSID[SID_HVBat_Volt1]							=	[["x07","7BB.2105.*;3,4,83"]	,	["x09","7BB.2141.*;3,2,29"]		,	["x10","7BB.2141.*;3,2,125"]	,	["x38","7DA.2134.*;3,2,125"]	,	["x61","7BB.2141.*;3,2,125"]	,	["x61Ph2","7BB.2141.*;3,2,125"]		]
:tabSID[SID_HVBat_Volt2]							=	[["x07",""]						,	["x09",""]						,	["x10","7BB.2142.*;3,2,69"]		,	["x38","7DA.2135.*;3,2,69"]		,	["x61","7BB.2142.*;3,2,69"]		,	["x61Ph2","7BB.2142,*;3,2,69"]		]
:tabSID[SID_HVBAT_Volt_Min]							=	[["x07","7BB.2107.5.0"]			,	["x09","7BB.2103.15.0"]			,	["x10","7BB.2103.15.0"]			,	["x38","7BB.2103.15.0"]			,	["x61","7BB.2103.15.0"]			,	["x61Ph2","7BB.2103.15.0"]			]
:tabSID[SID_HVBAT_Volt_Max]							=	[["x07","7BB.2107.3.0"]			,	["x09","7BB.2103.13.0"]			,	["x10","7BB.2103.13.0"]			,	["x38","7BB.2103.13.0"]			,	["x61","7BB.2103.13.0"]			,	["x61Ph2","7BB.2103.13.0"]			]
:tabSID[SID_HVBAT_Volt_Sum1]						=	[["x07","7BB.2107.9.0"]			,	["x09","7BB.2142.71.0"]			,	["x10","7BB.2142.71.0"]			,	["x38","7DA.2135.71.0"]			,	["x61","7BB.2142.71.0"]			,	["x61Ph2","7BB.2142.71.0"]			]
:tabSID[SID_HVBAT_Volt_Sum2]						=	[["x07","7BB.2107.11.0"]		,	["x09","7BB.2142.73.0"]			,	["x10","7BB.2142.73.0"]			,	["x38",""]						,	["x61","7BB.2142.73.0"]			,	["x61Ph2","7BB.2142.73.0"]			]
:tabSID[SID_HVBAT_FullCap]							=	[["x07","7BB.2161.7.0"]			,	["x09","7BB.2161.7.0"]			,	["x10","7BB.2161.3.0-32"]		,	["x38","7BB.2161.3.0"]			,	["x61","7BB.2161.7.0"]			,	["x61Ph2","7BB.2161.3.0"]			]			
:tabSID[SID_HVBAT_Odometer]							=	[["x07","7BB.2161.12.0"]		,	["x09","7BB.2161.12.0"]			,	["x10","7BB.2161.12.0"]			,	["x38","7BB.2161.12.0"]			,	["x61","7B6.2161.12.0"]			,	["x61Ph2","7BB.2161.12.0"]			]
:tabSID[SID_HVBAT_Soc]								=	[["x07","7BB.2108.9.0"]			,	["x09","7BB.2103.25.0"]			,	["x10","7BB.2103.25.0"]			,	["x38","7BB.2103.25.0"]			,	["x61","7BB.2101.39.0"]			,	["x61Ph2","7BB.2103.25.0"]			]
:tabSID[SID_HVBAT_UserSoc]							=	[["x07","7BB.2101.19.0"]		,	["x09","7BB.2101.37.0"]			,	["x10","7BB.2101.37.0"]			,	["x38","7BB.2103.29.0"]			,	["x61","7BB.2103.29.0"]			,	["x61Ph2","7BB.2101.37.0"]			]
// stop ici
:tabSID[SID_HVBAT_AvailableEnergy]					=	[["x07",""]						,	["x09","763.2101.9.7"]			,	["x10","7EC.22320C.4.0"]		,	["x38",""]						,	["x61",""]						,	["x61Ph2",""]						]
:tabSID[SID_InstantConsumption]						=	[["x07",""]						,	["x09",""]						,	["x10","7EC.223478.4.0"]		,	["x38",""]						,	["x61",""]						,	["x61Ph2",""]						]
                                    			
                                    			
:tabSID[SID_HVBAT_Voltage]							=	[["x07","7EC.223203.4.0"]		,	["x09","7BB.2142.73.0"]			,	["x10","7EC.223203.4.0"]		,	["x38","7EC.223203.4.0"]		,	["x61","7EC.223203.4.0"]		,	["x61Ph2","7EC.223203.4.0"]			]
:tabSID[SID_HVBAT_Current]							=	[["x07","7EC.223204.4.0"]		,	["x09","763.2101.17.4"]			,	["x10","7EC.223204.4.0"]		,	["x38","7EC.223204.4.0"]		,	["x61","7EC.223204.4.0"]		,	["x61Ph2","7EC.223204.4.0"]			]
:tabSID[SID_HVBAT_AvChPWR]							=	[["x07","7EC.223444.4.0"]		,	["x09","7BB.2101.23.0"]			,	["x10","7EC.223444.4.0"]		,	["x38","7EC.223444.4.0"]		,	["x61","7EC.223444.4.0"]		,	["x61Ph2","7EC.223444.4.0"]			]
:tabSID[SID_HVBAT_MaxCharge]						=	[["x07","7BB.2101.43.0"]		,	["x09","7BB.2101.43.0"]			,	["x10","7BB.2101.43.0"]			,	["x38","7BB.2101.43.0"]			,	["x61","7BB.2101.49.0"]			,	["x61Ph2","7BB.2101.43.0"]			]			
                                    			
// Autonomie                    					                                                               
:tabSID[SID_RangeEstimate]							=	[["x07","7EC.223451.4.0"]		,	["x09","763.2107.52.0"]			,	["x10","7EC.223451.4.0"]		,	["x38","7EC.223451.4.0"]		,	["x61","7EC.223451.4.0"]		,	["x61Ph2","7EC.223451.4.0"]			]
[... cut ...]
:tabSID[SIDv_HVBAT_Temp_Avg]						=	[["*ALL","900.2111.1.0"]]   // 900.2108.1.0	

// Ecritures (test sur RAZ compteur entretien)
:tabSID[SID_Ecriture_Entretien]						=	[["x07",""]		,	["x09",""]			,	["x10","763.2E020275300168"]		,	["x38",""]		,	["x61",""]		,	["x61Ph2",""]			]
:tabSID[SID_Lecture_Entretien_Courant]				=	[["x07",""]		,	["x09",""]			,	["x10","763.220202"]		,	["x38",""]		,	["x61",""]		,	["x61Ph2",""]			]
:tabSID[SID_Lecture_Entretien_Init]					=	[["x07",""]		,	["x09",""]			,	["x10","763.220201"]		,	["x38",""]		,	["x61",""]		,	["x61Ph2",""]			]

FredLeudon, 2019-06-27:
Hi all

As I didn't find FreeFrames data on isoTP Frames, i decided to implement FreeFrames on CanZE v2 for iOS

I got the free frames definition by reading the can_v10_02_20110126T152836.xml file.

So, questions are coming.

- in AuxBattTechActivity, FreeFrame SID_AuxStatus (638.37) is used to know if aux battery have to be replaced.

- in the definitions I loaded, FreeFrame 638 returns : 

ATMA -> [500F06E00000]

SID[.0638.1.0]	Data[TractionInstantConsumption]	Value[0 kW]	Raw Value[0]
SID[.0638.2.0]	Data[VehicleAutonomyMin]	Value[14 km]	Raw Value[14]
SID[.0638.3.2]	Data[VehicleAutonomyMax]	Value[25 km]	Raw Value[25]
SID[.0638.5.0]	Data[AuxInstantConsumption]	Value[0 kW]	Raw Value[0]

Those values seems strange.

Any idea ?

yoh-there, 2019-06-27:
Fred,

A few things.

1) I have translated all but 9 fields from free to isotp. See _FieldsAlt.csv
2) I implemented the replace 12 volt status (37 length 3) in Android, but I think it is useless. My battery failed without any indication
3) I have traction at 0 length 8, autonomy min at position 8 length 10, max at 18 length 19 and instant at 32 length 5. Are these values strange? Only if the battery was pretty full! ;-)

FredLeudon, 2019-06-27:
Thanks, will have a look.

yoh-there, 2019-10-22:
Looks like this issue is going stale. I will close it if there is no further progress.
Here is how I see it (calling @fesh for his opinion):

For CanSee dongles
- it doesn't matter, it's fast as hell and very reliable either way
- free frames is the almost always the faster of the two (if there is a choice)
- if one is principled, free frames fundamentally do not mess with the car

For ELM dongles
- it is most probably more reliable and faster to use ISOTP (ATMA command is crap)
- it is not certain that all fields we use are available (though I did most in FieldsAlt)
- if all fields can be done, probably all shitty dongles will work

So the most logical conclusion would be to use Fields for CanSee and FieldsAlt for ELM, but personally, I am not very eager to do all the testing and (hopefully) finding ISOTP alternatives for the last couple of fields. **Help is needed!!**

I propose to add a setting called "Alternative operation", which will select FieldsAlt, but only when an ELM dongle is selected.

fesch, 2019-10-23:
... all this combined with the fact that we are neither able to produce enough nor cheap enough CanSee dongles, the "alternative operation" mode seams to be a fair option for CanZE users.

But as @yoh-there said, an internal and automatic translation of free frames towards ISOTP frames would be needed, but that takes times (and help, yes!).

yoh-there, 2019-10-23:
I am down to 6 fields. A few of them we can (imho) "let go" or let go if on ELM. 
MeanEffectiveTorque (needed!)
Coasting Torque - FrictionTorque (needed!)
TotalPotentialResistiveWheelsTorque AKA bluebar (needed!)
TimeToFull (only used in chargingtech. Could be removed imho)
Charging status (used in chargingtech, prediction and auxbat graph. Could be removed imho)
Heater setpoint (only used in climate graph. Could be removed imho)

Suggestions appreciated.

yoh-there, 2019-11-02:
This is done and in final testing. Tick "Use ISOTP fields" in the settings. At this moment required to compile the branch AltFieldTest.

When sufficiently tested I propose to remove all fields / graph lines that use TimeToFull, ChargingStatus and HeaterSetpoint. Also, the _AltFields.csv, _Fields.csv and a few other things need cleaning up before merging into the Development branch.

closed 2019-11-03 by yoh-there