fesch/CanZE #510: DIY dongle ESP32 for iOS

yoh-there, 2018-12-29:
Fred, did you implement the "BobDue" driver on iOS? I am asking because I am further working on the home build dongle based on the ESP32, and I am considering adding "telnet" support in it for iOS.

It would create a local access point with DHCP in the 192.168.4.0/24 segment, and expose port 35000. It would accept only one session at a time.

FredLeudon, 2018-12-29:
This is all ready what I do with WiFi dongle. I open a socket on port 35000.

No job for me today ;-)

Regards 

Frédéric RICHARD

yoh-there, 2018-12-29:
Haha, yes, I carefully lined up the hard core communication (socket) with an ELM dongle. However, I am not talking about that part. The ELM driver is quite complex in fiddling with the ELM dongle. In the Android code base there is an alternative driver called BobDue. It uses basically two high level commands  (get free frame, get ISO-TP message), which is handled by the dongle. If you also implemented the BobDue driver, well, the ESP32 dongle is basically "iOS ready", see what I wrote on the CanZE blog ;-)

FredLeudon, 2018-12-29:
I didn’t exactly use this driver on iOS. I just use sockets.

If possible, I would like you to lend me a dongle to make some tests. 

Regards 

Frédéric RICHARD

yoh-there, 2018-12-29:
Hi Fred, I would love to but at the moment it is a self-build using my only ESP32 development board. I will order another ESP32, but it usually takes many weeks, especially around the Chinese new year to arrive. 

But again, it is totally fine to use sockets, that is EXACTLY what I do. We just don't use the complicated AT command set, but more something like

g1ca

to get free packet with ID 1ca, or

i7bc,2223

to send command 2223 to the ECU that answers on ID 7bc and return that answer. No ATSH, No ATMx, no ISO-TP decoding. It is all done in the ESP32 microcontroller. The BobDue driver simply issues above commands to the dongle, and yes over socket 35000 :-)

yoh-there, 2019-01-03:
Hi Fred,

1) There is more but you only need gxxx to get a free frame (xxx is the id) and ixxx,yyyy to get an ISO frame (xxx is the id, and yyyy is the PID. 

2) Yes/no. I would not go for 2 ESP32's. The ESP32 only has one CANbus controller, so we can't address them simultaneously. But I think (as in, I will check this) one can route CanRx and CanTx to any pin of the module. If true, you'd have to have two CANbus tranceivers. And of course a command in the firmware. Both would be easy. And enabling Android Auto would be easy, hehe! Great idea.

fesch, 2019-01-03:
Hi ...
@2: I just checked the online doc: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/can.html

> ... signal lines can be routed through the GPIO Matrix to GPIOs.

As far as I understood, there is only one CAN controller inside, so we can't listen on the two buses at the same time. Switching would be possible, but I think that will be far to slow!

yoh-there, 2019-01-03:
True if you want to "run" on the two buses simultaneously. But if you want to make say a "radio screen", well, why not?

Implementation we need to think about to maintain some compatibility. I think it would be wise to either add a bus number in the frame definition in CanZE, or, alternatively use high bits in the id (i.e. 0x1074e means 74e on the alternative bus). Dirty but effective. Such scheme could be implemented into CanSee too, so either create a new "switch bus" command, or implicitly do that using high bits.

fesch, 2019-01-03:
Yes, doing it that way would be possible.

Using a high bit would have the advantage, that the dongle itself asures the switch, but would be less generic meaning that the PID's need some adjustment. Anyway, adjustments would be needed ...

Still I think that adding a second transceiver would be worth trying it.

But do we have enough documentation of the MM-bus?

yoh-there, 2019-01-03:
I have a "secret spreadsheet". Of course we would have to convert that to a new / added "fields" initialisation........

FredLeudon, 2019-01-04:
Using DDT2000 database, it's easy to know which ECU uses which Can bus. That what I do in CanZE for iOS.

if ESP32 have only one CAN controller, why not using 2 ESP32, one with Wifi and one with Bluetooth ?

fesch, 2019-01-04:
@FredLeudon 
I see different problems with the solution using 2 ESP's:

1. Two different firmware's are needed
2. Any application will need to use two different channels to communicate with each ESP
3. Two ESP's take twice as much space in the dongle as a single one does ;-)

yoh-there, 2019-01-06:
Before going deeper in design issues, let's step back for a minute and determine WHY we might want to do this. I guess the most important driver is: do we want to switch / query the to buses real time / full speed. If not, the obvious answer of course is simply use one ESP32 with two transceivers.

As far as I know there is not much useful real time info on the MM bus. GPS location is not on the free frames (though it might be on an ISO-TP query to R-LINK. @FredLeudon have you checked that with DDT4All? If so, do you have query definition for me please?). For the current version of CanZE the most useful data would be the remaining distance from the Tomtom route planner. That also is not on the free frames, so same question to Fred on that. Now suppose those can be done ISO-TP, would switching buses every 10 or 20 seconds be a problem? I don't think so, but that is my opinion.

Is hooking up to the MM bus a good plan for non-real time things? I think yes. At minimum we could add an "Enable Android Auto" command and get Firmware versions.

yoh-there, 2019-01-08:
Dual can tranceivers is implemented in the CanSee development branch, see it's readme.md for description. I used the high bit approach (0x40000yyy) as id selects bus 2, 0xyyy selects bus 1, leaving space in the id for 29 it implementation.

I won't close this issues as we've not been able to test on iOS yet.

yoh-there, 2019-01-13:
@FredLeudon Thanks for the data. Unfortunately I could not find (but maybe I overlooked) navigation "distance to destination", neither the GPS position. No problem, but it makes any fast bus switching unneccessary. I think we should decide to stick with one ESP and optionally two transceivers.
Since you raised the issue, your feedback is much appreciated :-)

FredLeudon, 2019-01-14:
Hi,

There is what we’re doing. Using one ESP and two can bus transceivers. 
CANSee code is modified to allow selecting which transceiver to use. 
I already received ESP, DC-DC converter. I’m waiting for transceivers and OBB plug.

You’re right, there are no GPS neither Distance to destination available on navigation ECU, but being able to address navigation or radio with the same dongle as other ECUs is still useful.

It would be great to add a Android-Auto activation for example, or being able to retrieve DTC for all ECUs at a time.

On IOS version (4.0, still in beta), the way I connect to a car is :

• establish data connection to the dongle (BT or WiFi and CANSee when my diy dongle will be operational)
• querying ECUs to read vehicle’s VIN
• once VIN is read, I can load ECU address array*
• each address is scanned to retrieve ECU version numbers*
• ecu definition is downloaded from my server*
• these definition is used to load Request, data items and data arrays.

*operations are done only once and ecus version are stored locally in database files.

It works fine (some data are missing) for twizy, kangoo and Zoe. 
I have to test on Zoe Q/R90 and Q/R110 and Fluence. 
I would like to test it too on upgraded Zoe. Somebody I know will upgrade this year (I hope) and I will be able to test it too.

Frédéric RICHARD

yoh-there, 2019-01-14:
Understood, thank you. It would be nice if you could share the code you describe. Seems like we're doing a lot of research in parallel.

Regarding the way of loading definitions I propose to continue that at issue #504 as it's unrelated to this one.

We'll further implement the dual bus and long ISOTP sending in CanSee, discussion here https://gitlab.com/jeroenmeijer/cansee/issues/1

Nothing further to discuss on this issue, I am closing it.

closed 2019-01-14 by yoh-there