fesch/CanZE #515: Enhance start-Stop poller behavior

yoh-there, 2019-01-27:
For some activities (ELM, DTCs, TPMS) we stop the poller and restart it when done. IMHO this is all a bit tedious, uses slightly unlogical names (Init - are we restarting the dongle settings? Join - what is join?) and should be improved, not in the least because on one hand we sometimes need to send keepalive messages (ideal work for the poller), but on the other hand long message sending / receiving in the foreground should not be interrupted by an active poller background thread.

My proposed solution is to add a public method `Message injectRequest (Frame f)` to the device class. It should pause the poller as soon as it has finished it's current operation (if any), query the frame, continue the poller, then return a message. This method should not clear the queues, just ensure things are silent while running.

Another way of achieving the same result is to inject the frame to the top of the queue, unsleep it if that would be needed/possible, and remove it again when done. This however would require some code to synchronize the result.

The CanZE activity can then i.e. inject a Diagnostic Tool Present frame, add the keepalive field tot the poller queue, then read ECU values using inject again as needed. Nice side effect is that the application queue will remain active.

fesch, 2019-01-27:
to join someone = fix an appointment with some one, wait for him to finish his work and wait until he reaches the meeting point

init >> yes, the device is restarted as for the first time

yoh-there, 2019-01-27:
I hear you :-) I don't want to joining anything. A better analogy would be me shouting "hang up the phone and get me that file, then you can babble on!!" :-P . I am not criticizing the naming, I am just making the case for a (simple) solution for the use case. I am pretty sure we'll need it more in the future.

TPMS code now is ugly. Re-initialization of the device is rather overkill for what I am trying to accomplish, especially on those blasted ELMs.

I have no problem giving a stab an implementing it, but I like to start with a decent discussion about being on the right track or not :-)

yoh-there, 2019-01-27:
OK, I get it better now. The Stop part just sets a variable telling the poller thread to end itself. Cool. My only string suggestion in that part is the 5 seconds sleeping. This could lead to silly long wait times for.......

..... Joining, which is in essence a synchronous waiting for that thread to die.

So while my proposed strategy is to "pause" the thread (suspend is deprecated), the join is to simply let it die and create a new one when needed. While it "sounds" wasteful, it's fine and saves ugly synchronization / thread communication.

So, I will now rephrase my suggestion and add a question.
1) Implement logical name sin the Device class, such as for example "pollerStart" and "pollerStop". Basically this is a suggestion to refactor, and to me not a big deal if we do it or not.
2.a) pollerStart should have a parameter telling it if it needs to initialize the physical device or, if it can assume that it has been successfully initialized.
2.b) Alternatively, and probably preferably, we should optimize the code so "pollerStart" will by itself know the device is in an initialized state.

I realize (correct I hope), that killing and restarting the poller thread does not mess with the request queues at all

Now the question:
We now check if the device is null for restarting. That doesn't make a lot of sense. If device is null, the code would have crashed long before. So I think we should
A) enclose the entire code block in  a check, like
`if (device != null) {
   device.pollerStop();
   doMyThing();
   device.pollerStart();
}`, or
B) take off those checks; just crash if no device is set, or
C) mainActivity should allow NOTHING if device is null after loadSettings.
I propose use B as it won't be a problem anymore when we do C.

fesch, 2019-01-27:
I agree ;)

I've implemented two new methods which do the same you are actually doing, so they should work out of the box. After that step, I'll check to define them in order to increase performance.

yoh-there, 2019-01-28:
Did I miss a commit/push, or was that not what you meant?

yoh-there, 2019-02-03:
Added needed changes. CLosing

closed 2019-02-03 by yoh-there