fesch/CanZE #521: HV PTC Battery Conditioning Level [Request]

matthew1471, 2019-02-02:
Possible to include IH_HVPTCConsumption to read what level of High Voltage battery heating is taking place?

Should be 7 bits with the unit being "W" in steps of 50. First byte 25, bit offset 1 according to the X10_X98_X87_X52ClimBoxGen1_v08_3_20160722T152536.xml file I found.

Tries to work out from the source how you get IH_HVBatConditionningMode with "432.36" and also couldn't work out the CSV file's importance in the asset folder?

CL_PTCType would also be useful and usage field could disappear if no PTC.

yoh-there, 2019-02-02:
Hi Matthew. The only csv to look at is _Fields. The others are only used for DTCs and dumoing entire ECU contents (which is somthing I am  sure YOU want to do.

The Renault XMLs are not easy to work with as the there is an indirection for the bit positioning. I have a program stitching it all together to [id].[request].[startbit], which is how we address ISO-TP fields in CanCE. All ID's below 0x700 are "free floating" frames, basically the hard internals of the car.

matthew1471, 2019-02-03:
I have found the relevant bits in the Renault docs (looks like the EVC knows about it), will check the CSVs next

```
<Data Name="($3498) This flag shows the battery relay PTC n°1(X10,L38G2,X61G3) request">
<Comment><![CDATA[B_BattCTP_relay1_req]]></Comment>
<Mnemonic>DID_$3498</Mnemonic>
<Bits count="1">
<List>
<Item Value="0" Text="no activation"/>
<Item Value="1" Text="activation"/></List>
</Bits>
</Data>
..
<Data Name="($349A) This flag shows the battery relay PTC n°2 (X10,L38G2,X61G3) request">
<Comment><![CDATA[B_BattCTP_relay2_req]]></Comment>
<Mnemonic>DID_$349A</Mnemonic>
<Bits count="1">
<List>
<Item Value="0" Text="no activation"/>
<Item Value="1" Text="activation"/></List>
</Bits>
</Data>
..
<Data Name="($349C) This flag shows the battery relay PTC n°3 (X10,L38G2,X61G3) request">
<Comment><![CDATA[B_BattCTP_relay3_req]]></Comment>
<Mnemonic>DID_$349C</Mnemonic>
<Bits count="1">
<List>
<Item Value="0" Text="no activation"/>
<Item Value="1" Text="activation"/></List>
</Bits>
</Data>
```

yoh-there, 2019-02-03:
I can add this is in the CLIMA screen

matthew1471, 2019-03-03:
Am always seeing:

PTC Relay 1 On
PTC Relay 2 On
PTC Relay 3 On

Even when cells are around 27/28oC. Limitation of the field we're querying (perhaps PTC isn't controlled in the correct place on my model)? or labels wrong way around?

Will do more testing when the batteries are colder.. to see whether the status is just the wrong way around.

yoh-there, 2019-03-03:
Same for me actually. I don't think this is very useful, unless it's model dependent. I'd be interested in your findings.

matthew1471, 2021-01-07:
I think the PTC relay values were slightly off because All Data was showing correct states:

7ec.623498.31 This flag shows the battery relay PTC n°1(X10:L38G2
7ec.623499.31 relay thermoplunger (X07) n°1 request:0:no activation
7ec.62349a.31 This flag shows the battery relay PTC n°2 (X10:L38G2
7ec.62349b.31 relay thermoplunger (X07) n°2 request:0:no activation
7ec.62349c.31 This flag shows the battery relay PTC n°3 (X10:L38G2
7ec.62349d.31 relay thermoplunger (X07) n°3 request:0:no activation

Note the relay request is 623499, 62349b and 62349d.. not 623498, 62349a, 62349c.

yoh-there, 2021-01-07:
I got a completely different message in my mail for your comment @matthew1471 I assume that's superseded?

As for your comment: more OTC's is certainly correct. AFAIK there is a small 12V one for the windshield for non-nordic and a 400V one for Nordic, at least for the older models.

AS you stated in speakev the battery has an electrical heater, except 22 kWh non-Nordic

But I am fairly sure the SIDs you mention are not the PTCs. They read like the valves to open/close the feeding of the evaporators. Maybe that is what you meant. Loud little buggers by the way!

matthew1471, 2021-01-07:
I kept changing it then correcting myself when I realised I was wrong :(.

The "battery HVAC" I think is where I want to be looking to work out when it's activated and not. I thought maybe evaporator might be a good shot as that's what the rear HVAC (battery HVAC) PDF talks about.

I've got a CanZE "All Data" for the EVC and HVAC and I think probably the sensiblest thing for me to do will be to get the battery conditioning to finish then take another snapshot and diff my 2 files. Most of my 2019 CanZE All Data has changed fields and formats enough that I think I need to take a fresh one. I am keen to hunt down how aggressive we're battery heating.

WIth ZE50 I am starting to think there might not be a heater in the battery HVAC.. Also there's a more prominent clicking coming from the dash off and on when the battery should be warming.

yoh-there, 2021-01-07:
Got it. But I think you're looking possibly in the wrong direction. The HVAC stuff is cooling only. The heat pump does not do any heating. I really doubt the HVAC computer does anything else than switching the valves as far as the battery is concerned. The EVC is probably a better candidate.

If I were you I would be watching for differences especially when it is on scheduled charging, as that makes sense to do predicted battery heating.

To get back to CanZE, is there anything you are requesting or reporting? Such as these HVAC valves to display?

matthew1471, 2021-01-08:
2 CanZE "All Data"'s for EVC:

Amongst some other values that basically say the same thing:

When the battery HVAC is heating and not charging (and presumably similar for cooling):
>7ec.623494.24,($3494) Instantaneous DCDC output power,350,W
>7ec.6234a4.29,($34A4) Battery cooling order,4:Warm_Bat_Int_CTP,
>7ec.6234a6.25,($34A6) Battery HVAC speed rotation request (command of the blower),40,%

When the battery HVAC is not heating but charging on 99% (and presumably same for not cooling):
>7ec.623494.24,($3494) Instantaneous DCDC output power,180,W
>7ec.6234a4.29,($34A4) Battery cooling order,0:No_Cool_Bat,
>7ec.6234a6.25,($34A6) Battery HVAC speed rotation request (command of the blower),0,%

I can give you a full copy of both files if you want to see the before/after. It's interesting because a lot of the cooling flags get tripped for heating too... I think they're more general battery HVAC flags.

Unfortunately still not haven't quite yet found the different relays. The ask of CanZE is to show which of the 3 states the battery heating is happening.. but I'm not entirely sure that's accessible from CAN now.. I think it may have to be inferred from Battery HVAC fan speed.

The battery HVAC (different from car HVAC) is under the boot and heard from rear wheel arches.. I'd missed that detail in the Google Docs link but it makes sense now.. that's why the rear of the car has a fan noise and not the front!

I don't think scheduled charging has any impact on battery heating.. I can trip battery heating on a car that's on 100% charged by just plugging it in when the cells are cold.

I'm guessing you don't see the **Battery HVAC** on CAN as a separate ECU?

matthew1471, 2021-01-08:
There's some stuff in HVAC but agree it's not terribly useful:

>764.6144.133,IH_HVBatConditionningMode,3:Conditionning not requested,
>764.6144.191,OH_BatteryBlowerState,0:Not active,

and 

>764.6144.133,IH_HVBatConditionningMode,2:Heating conditionning requested,
>764.6144.191,OH_BatteryBlowerState,1:Active,