fesch/CanZE #523: "Battery kilometers" and subtle rounding difference?

matthew1471, 2019-02-02:
In the strings there is

`<string name="label_HKM">Battery kilometers</string>`

This is confusing if you have CanZE set to show miles. Can I please suggest "Battery Odometer"? or that the label alters to say "Battery Mileage" if "Miles" is set.

Also I think there might be a rounding error sometimes? My car sometimes reports 1 mile less than CanZE is reporting.. so I suspect there to be an extremely subtle rounding issue.

matthew1471, 2019-02-03:
Or maybe just "Battery Mileage" is appropriate for both units? https://www.quora.com/The-distance-on-a-cars-odometer-is-called-mileage-Is-there-an-equivalent-word-if-its-a-car-with-metric-units-km

matthew1471, 2019-02-03:
[image]

Car dash shows 10784 miles.

yoh-there, 2019-02-03:
Fixed commit 76fd909

closed 2019-02-03 by yoh-there

matthew1471, 2019-03-03:
Fine by me about rounding, just wanted to make sure we knew about it and had a reason as to why it was deemed correct if it was correct. Thanks for changing the label :).

commit 76fd9095 (2019-02-03): Fixed #523