fesch/CanZE #524: Screen Rotation Kills Bluetooth Connection

matthew1471, 2019-02-03:
I've observed that when you rotate the screen CanZE will drop the bluetooth connection (bluetooth light goes off on dongle) and CanZE reports connection drop.

I believe this is due to the way that Android "activites" are resumed/redrawn and CanZE incorrectly thinking the application has been suspended and resumed.

This is on Android 9 and by tapping the manual rotate button.

yoh-there, 2019-02-09:
I was wrong about the 10 minutes. A valid point but after discussing, we won't fix this as it is / should not hampering the app except a bit of delay when rotating. Marking as such and closing.

closed 2019-02-09 by yoh-there

matthew1471, 2019-03-03:
It seemed to cause quite a long delay on my phone and more than anything else it was how unexpected it was "oh the dongle has lost connection again... why? This connection is super un-reliable.. Oh hang on I wonder if it's something to do with I just turned my phone on its side?"