fesch/CanZE #525: "Max battery charge/regen kW" and "DC Power kW" Wrong Way Around?

matthew1471, 2019-02-03:
Possible I've mis-understood the values but are they the wrong way around?

During this cold weather I'm doing a lot of battery PTC conditioning and I'm observing the Max battery charge/regen kW is often higher than the DC Power kW

[image]

If DC Power kW is the total in then surely the amount going to the battery specifically for charging should be less?

matthew1471, 2019-02-03:
Even happens when pressing the pre-condition button and the charge post is flashing to say that there's no charge instruction yet... you see -1.2kW coming from the battery but 0.6 showing as "Max battery charge/regen kW"

[image]

Then once the charger had started supplying more energy the max power increased as I would expect the DC to have done. The DC had a smaller value to charge back up the charge I had just lost from pre-conditioning from battery while the car was still waking up and negotiating with the charger.

Merlin2001, 2019-02-03:
Hi Matthew,

only an end user here, too.

But I think your values are completely valid.
*Max battery charge/regen kW* should be the maximum power, your battery _could_ currently be charged at (according to its current condition, i.e. SoC and temperature) - meaning that _actual_ charge power must be below this value.

Now *DC Power kW* should give you the _actual_ (DC-side) power that is put into or consumed from the battery. It must thus be lower than the maximum mentioned above.

Hope this helps :)
Marcus

matthew1471, 2019-03-03:
I stand corrected and defer to your greater experience then :)